intel/ros2_intel_realsense #5: Failed to load entry point 'status': librmw.so when running the ROS2 launch file

rosprog1, 2018-05-25:
Required Info |  
-- | --
Camera Model | D415
Firmware Version | 05.09.11.00
Operating System & Version | Ubuntu 16.04
Kernel Version (Linux Only) | 4.13.0-43-generic
Platform | PC
SDK Version | 2.11.1

**Issue Description**
Follow below instructions to build, but it's failed to launch as you can see below error. any idea to resolve this?
https://github.com/intel/ros2_intel_realsense
$ ros2 run realsense_ros2_camera realsense_ros2_camera
Failed to load entry point 'status': librmw.so: cannot open shared object file: No such file or directory
The C extension '/opt/ros/ardent/lib/python3.5/site-packages/rclpy/_rclpy.cpython-35m-x86_64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'stop': librmw.so: cannot open shared object file: No such file or directory
The C extension '/opt/ros/ardent/lib/python3.5/site-packages/rclpy/_rclpy.cpython-35m-x86_64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions

sharronliu, 2018-05-25:
the build log says "Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions", would you let me know if this solve the problem?

sharronliu, 2018-11-02:
Closing this since no more questions from the submitter.

closed 2018-11-02 by sharronliu