intel/ros2_intel_realsense #8: port aligned depth images from intel-ros/realsense

yechun1, 2018-07-30:
@dorodnic @sharronliu 
We are developing [ros2_object_analytics](https://github.com/intel/ros2_object_analytics) which use [intel/ros2_intel_realsense](https://github.com/intel/ros2_intel_realsense) as dependence.We require a feature "publish aligned depth images" from realsense package. As we checked [intel-ros/realsense 2.0.2](https://github.com/intel-ros/realsense/tree/2.0.2) had already supported publishing aligned frames feature. But ros2_intel_realsense is old, and just ported based on https://github.com/intel-ros/realsense/tree/2.0.1, and have no any updates. May I ask if you have any plan to port latest intel-ros/realsense code to ros2_intel_realsense?  I would also like to contribute on porting 2.0.2 (enable aligned depth images) from realsense to ros2 realsense. Do you have any suggestion? Thanks.

sharronliu, 2018-07-31:
It will be great to keep enabling v2.0.2 features in ROS2. Yet not started due to my bandwidth unavailable in short while. Contribution is welcomed anytime :).

sharronliu, 2018-11-02:
https://github.com/intel/ros2_intel_realsense/pull/11

closed 2018-11-02 by sharronliu