intel/ros2_intel_realsense #20: rviz2 error after realsense_ros2_camera

yahoo2016, 2018-12-23:
I installed ROS2 Crystal, librealsense2 and realsense_ros2_camera on Jetson TX2.
realsense_ros2_camera was fine:
[INFO] [RealSenseCameraNode]: setupStreams...
[INFO] [RealSenseCameraNode]: depth stream is enabled - width: 640, height: 480, fps: 30
[INFO] [RealSenseCameraNode]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[INFO] [RealSenseCameraNode]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[INFO] [RealSenseCameraNode]: color stream is enabled - width: 640, height: 480, fps: 30
[INFO] [RealSenseCameraNode]: publishStaticTransforms...
[INFO] [RealSenseCameraNode]: RealSense Node Is Up!
but rviz2 won't display video and generated the following errors:
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [rviz2]: Stereo is NOT SUPPORTED

yechun1, 2018-12-24:
seems the realsense is work normally. you could use ros2 topic to check the topics.
For your issues, have you added image and choose topic such as /camera/color/image_raw in RVIZ2?

yahoo2016, 2018-12-24:
It worked when I tried to add images "By topic" and selected the published topics.
TH\hanks!