intel/ros2_intel_realsense #24: Build Error - Intel RealSense SDK 2.0 is missing

lbegani, 2019-01-29:
`lkumarbe@intel:~/ros2_ws/src$ git clone https://github.com/intel/ros2_intel_realsense.git`

> Cloning into 'ros2_intel_realsense'...
remote: Enumerating objects: 86, done.
remote: Counting objects: 100% (86/86), done.
remote: Compressing objects: 100% (43/43), done.
remote: Total 231 (delta 47), reused 75 (delta 43), pack-reused 145
Receiving objects: 100% (231/231), 933.45 KiB | 498.00 KiB/s, done.
Resolving deltas: 100% (107/107), done.

`lkumarbe@intel:~/ros2_ws/src$ ls`
> ament     image_common  ros   ros2_intel_movidius_ncs  ros2_object_analytics  ros-perception  ros-visualization
eProsima  osrf          ros2  ros2_intel_realsense     ros2_object_msgs       ros-planning    vision_opencv

`lkumarbe@intel:~/ros2_ws/src$ cd ..`
`lkumarbe@intel:~/ros2_ws$ source install/local_setup.bash`
`lkumarbe@intel:~/ros2_ws$ colcon build --base-paths src/ros2_intel_realsense`
> Starting >>> realsense_camera_msgs
Finished <<< realsense_camera_msgs [5.67s]                       
Starting >>> realsense_ros2_camera
--- stderr: realsense_ros2_camera                         
CMake Warning at CMakeLists.txt:52 (find_package):
  By not providing "Findlibrealsense2.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "librealsense2", but CMake did not find one.
  Could not find a package configuration file provided by "librealsense2"
  with any of the following names:
    librealsense2Config.cmake
    librealsense2-config.cmake
  Add the installation prefix of "librealsense2" to CMAKE_PREFIX_PATH or set
  "librealsense2_DIR" to a directory containing one of the above files.  If
  "librealsense2" provides a separate development package or SDK, be sure it
  has been installed.
CMake Error at CMakeLists.txt:54 (message):
   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases ---
Failed   <<< realsense_ros2_camera	[ Exited with code 1 ]
Summary: 1 package finished [5.82s]
  1 package failed: realsense_ros2_camera
  1 package had stderr output: realsense_ros2_camera

I have installed librealsense2 SDK from debian following the instructions from [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)

`lkumarbe@intel:~/ros2_ws$ modinfo uvcvideo | grep "version:"`
> version:        1.1.2.realsense-1.3.4
srcversion:     C4FB5E9B212E9C39ED6CD7C

yechun1, 2019-01-30:
We have rebuild librealsense for ros2 users, would you please try to install ros-crystal-librealsense2 instead?

lbegani, 2019-01-31:
After installation of `ros-crystal-librealsense2` , the error is fixed. Thanks

closed 2019-01-31 by lbegani

simutisernestas, 2020-07-16:
I'm having same exact problem with foxy and there is no prebuilt package. Do you plan to add it? Any tips how to solve it in a different way?

zhuhang0796, 2020-08-18:
> ros-crystal-librealsense2

How to install ,could you please reply more details?