intel/ros2_intel_realsense #25: Errors when pressed Ctrl+C

lbegani, 2019-01-31:
To reproduce the error - 
1. `$ ros2 launch realsense_ros2_camera ros2_intel_realsense.launch.py`
2. While the node is running, hit Ctrl+C

>  [rcutils|error_handling.c:106] rcutils_set_error_state()
This error state is being overwritten:
  'rcl node's context is invalid, at /home/lkumarbe/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:531'
with this new error message:
  'publisher's context is invalid, at /home/lkumarbe/ros2_ws/src/ros2/rcl/rcl/src/rcl/publisher.c:313'
rcutils_reset_error() should be called after error handling to avoid this.
<<<

mkhansen-intel, 2019-02-08:
I see the same issue:
```
[INFO] [RealSenseCameraNode]: RealSense Node Is Up!

^CInterruptSignal is received! Terminate RealSense Node...
 08/02 11:09:22,672 ERROR [139724605875968] (source.cpp:127) Exception was thrown during user callback!

>>> [rcutils|error_handling.c:106] rcutils_set_error_state()
This error state is being overwritten:

  'rcl node's context is invalid, at /home/robot/rap_framework-amr_sdk_crystal/src/amr/ros2/core/ros2/rcl/rcl/src/rcl/node.c:531'

with this new error message:

  'publisher's context is invalid, at /home/robot/rap_framework-amr_sdk_crystal/src/amr/ros2/core/ros2/rcl/rcl/src/rcl/publisher.c:313'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
```

ahuizxc, 2019-02-14:
> I see the same issue:
>
> ```
> [INFO] [RealSenseCameraNode]: RealSense Node Is Up!
>
>
> ^CInterruptSignal is received! Terminate RealSense Node...
>  08/02 11:09:22,672 ERROR [139724605875968] [...]
https://github.com/intel/ros2_intel_realsense/pull/30

closed 2019-07-10 by ahuizxc