intel/ros2_intel_realsense #27: enable_pointcloud=True results in segfault when running with D415

mkhansen-intel, 2019-02-08:


sharronliu, 2019-02-12:
PointCloud should be supported for D415.

sharronliu, 2019-02-12:
https://github.com/intel/ros2_intel_realsense/pull/28

ahuizxc, 2019-02-13:
> When I run with the D415 camera and enable_pointcloud=True, which is the default I get a crash / segfault with no error message:
> 
> ```
> robot@nuc-i7:~$ ros2 run realsense_ros2_camera realsense_ros2_camera
> [INFO] [RealSenseCameraNode]: RealSense ROS v2. 0.1
> [INFO] [RealSenseCameraNode]: Running with LibRealSense v2.16.5
> [INFO] [RealSenseCameraNode]: getParameters...
> [INFO] [RealSenseCameraNode]: setupDevice...
> [INFO] [RealSenseCameraNode]: Device Name: Intel RealSense D415
> [INFO] [RealSenseCameraNode]: Device Serial No: 725112060242
> [INFO] [RealSenseCameraNode]: Device FW version: 05.11.01.00
> [INFO] [RealSenseCameraNode]: Device Product ID: 0AD3
> [INFO] [RealSenseCameraNode]: Sync Mode: On
> [INFO] [RealSenseCameraNode]: Device Sensors: 
> [INFO] [RealSenseCameraNode]: Stereo Module was found.
> [INFO] [RealSenseCameraNode]: RGB Camera was found.
> [INFO] [RealSenseCameraNode]: Fisheye sensor isn't supported by current device! -- Skipping...
> [INFO] [RealSenseCameraNode]: Gyro sensor isn't supported by current device! -- Skipping...
> [INFO] [RealSenseCameraNode]: Accel sensor isn't supported by current device! -- Skipping...
> [INFO] [RealSenseCameraNode]: setupPublishers...
> [INFO] [RealSenseCameraNode]: setupStreams...
> [INFO] [RealSenseCameraNode]: depth stream is enabled - width: 640, height: 480, fps: 30
> [INFO] [RealSenseCameraNode]: infra1 stream is enabled - width: 640, height: 480, fps: 30
> [INFO] [RealSenseCameraNode]: infra2 stream is enabled - width: 640, height: 480, fps: 30
> [INFO] [RealSenseCameraNode]: color stream is enabled - width: 640, height: 480, fps: 30
> robot@nuc-i7:~$ 
> ```
> 
> No Error, but camera starts then quits!
> 
> When I run with enable_pointcloud=False:
> 
> ```
> robot@nuc-i7:~/rap_framework-amr_sdk_crystal$ cat realsense_params.yaml 
> RealSenseCameraNode:
>   ros__parameters:
>     enable_depth: True
>     enable_infra1: True
>     enable_infra2: True
>     enable_pointcloud: False
> robot@nuc-i7:~/rap_framework-amr_sdk_crystal$ ros2 run realsense_ros2_camera realsense_ros2_camera __params:=realsense_params.yaml 
> [INFO] [RealSenseCameraNode]: RealSense ROS v2. 0.1
> [INFO] [RealSenseCameraNode]: Running with LibRealSense v2.16.5
> [INFO] [RealSenseCameraNode]: getParameters...
> [INFO] [RealSenseCameraNode]: setupDevice...
> [INFO] [RealSenseCameraNode]: Device Name: Intel RealSense D415
> [INFO] [RealSenseCameraNode]: Device Serial No: 725112060242
> [INFO] [RealSenseCameraNode]: Device FW version: 05.11.01.00
> [INFO] [RealSenseCameraNode]: Device Product ID: 0AD3
> [INFO] [RealSenseCameraNode]: Sync Mode: On
> [INFO] [RealSenseCameraNode]: Device Sensors: 
> [INFO] [RealSenseCameraNode]: Stereo Module was found.
> [INFO] [RealSenseCameraNode]: RGB Camera was found.
> [INFO] [RealSenseCameraNode]: Fisheye sensor isn't supported by current device! -- Skipping...
> [INFO] [RealSenseCameraNode]: Gyro sensor isn't supported by current device! -- Skipping...
> [INFO] [RealSenseCameraNode]: Accel sensor isn't supported by current device! -- Skipping...
> [INFO] [RealSenseCameraNode]: setupPublishers...
> [INFO] [RealSenseCameraNode]: setupStreams...
> [INFO] [RealSenseCameraNode]: depth stream is enabled - width: 640, height: 480, fps: 30
> [INFO] [RealSenseCameraNode]: infra1 stream is enabled - width: 640, height: 480, fps: 30
> [INFO] [RealSenseCameraNode]: infra2 stream is enabled - width: 640, height: 480, fps: 30
> [INFO] [RealSenseCameraNode]: color stream is enabled - width: 640, height: 480, fps: 30
> [INFO] [RealSenseCameraNode]: publishStaticTransforms...
> [INFO] [RealSenseCameraNode]: RealSense Node Is Up!
> ```
> 
> That works fine.
> When I ran gdb and did a backtrace to find the failure, it is in this line of code:
> 
> [ros2_intel_realsense/realsense_ros2_camera/src/realsense_camera_node.cpp](https://github.com/intel/ros2_intel_realsense/blob/b76e6e185d6a3a3b554292d19eeaa5a2a7a677ea/realsense_ros2_camera/src/realsense_camera_node.cpp#L523)
> 
> Line 523 in [b76e6e1](/intel/ros2_intel_realsense/commit/b76e6e185d6a3a3b554292d19eeaa5a2a7a677ea)
>  publishPCTopic(t); 
> 
> Not sure why it is segfaulting there.
> 
> When I run with the D435 camera, this issue doesn't happen. Is pointcloud supported for D415?

Hi matt, I also encountered the same problem when I use D415 camera, it caused by wrong camera parameters. Re-calibrating the camera and the problem will be solved :)

closed 2019-09-05 by ahuizxc