intel/ros2_intel_realsense #31: build failed - intel realsense sdk is mssing..

revizes, 2019-02-25:
I'm trying to build librealsense2 & realsense_ros2_camera.

whenever i build librealsense2 with local account, "permission denied" message about ld.so.cache is shown.
anyway, i ignored that message.

Problem is building realsense_ros2_camera.
If i install ros-crystal-librealsense2 package, build is ok.
but if i use librealsense2 from building source, i always see below message.
I built ros2 from source code. so i think i can't use debian package.

Please give me some advice.
Thanks.

--- stderr: realsense_ros2_camera
CMake Warning at CMakeLists.txt:53 (find_package):
  By not providing "Findlibrealsense2.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "librealsense2", but CMake did not find one.

  Could not find a package configuration file provided by "librealsense2"
  with any of the following names:

    librealsense2Config.cmake
    librealsense2-config.cmake

  Add the installation prefix of "librealsense2" to CMAKE_PREFIX_PATH or set
  "librealsense2_DIR" to a directory containing one of the above files.  If
  "librealsense2" provides a separate development package or SDK, be sure it
  has been installed.

CMake Error at CMakeLists.txt:55 (message):
  

  

   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases

  

---
Failed   <<< realsense_ros2_camera	[ Exited with code 1 ]

yechun1, 2019-02-27:
we enabled ros2debian branch as ROS2 package to build librealsense, would you please have a try?  https://github.com/IntelRealSense/librealsense/tree/ros2debian

cd ~/ros2_ws/src
git clone https://github.com/IntelRealSense/librealsense
git checkout ros2debian
cd ..
colcon build --base-paths src/librealsense

revizes, 2019-02-28:
@yechun1 : it worked.
Thank you.

closed 2019-02-28 by revizes