intel/ros2_intel_realsense #47: Not supported on Mac

bobgates, 2019-04-25:
Now that the main RealSense library is getting more mature on Mac, can we get the ros2 library also running on Mac? Specifically, the 2.21 version of librealsense supports the T265 on the Mac - how about on Ros2 on the Mac?

ahuizxc, 2019-06-14:
> Now that the main RealSense library is getting more mature on Mac, can we get the ros2 library also running on Mac? Specifically, the 2.21 version of librealsense supports the T265 on the Mac - how about on Ros2 on the Mac?

Hi OSX support is still ongoing, we will inform you as soon as it's done! thanks:)

ahuizxc, 2019-07-10:
> Now that the main RealSense library is getting more mature on Mac, can we get the ros2 library also running on Mac? Specifically, the 2.21 version of librealsense supports the T265 on the Mac - how about on Ros2 on the Mac?

Hi bob, due to cv_bridge has some dependency problems, so currently this wrapper does not support MacOS, thanks

closed 2019-07-10 by ahuizxc