intel/ros2_intel_realsense #59: 当realsense_ros2_camera和rviz2分开启动时，不会发布TF tree

MengNan-Li, 2019-07-04:
ros2 launch realsense_ros2_camera ros2_intel_realsense.launch.py
在launch中启动realsense_ros2_camera和rviz2是正常的，但是把launch中启动rviz2部分注释掉，只启动realsense_ros2_camera，手动启动Rviz2就不会有TF tree了。因为我需要配置camera和机器人底盘的坐标关系，没有camera各个镜头的TF tree是很麻烦的。

ahuizxc, 2019-07-10:
wrapper中的tf是静态tf变换 只在启动节点的时候发布一次（可能是ros2的bug）你可以尝试先启动机器人监听camera的topic和tf变换 然后再启动camera就可以了

> ros2 launch realsense_ros2_camera ros2_intel_realsense.launch.py
> 在launch中启动realsense_ros2_camera和rviz2是正常的，但是把launch中启动rviz2部分注释掉，只启动realsense_ros2_camera，手动启动Rviz2就不会有TF tree了。因为我需要配置camera和机器人底盘的坐标关系，没有camera各个镜头的TF tree是很麻烦的。

the tf transform in this wrapper is static_tf_broadcaster, so it only broadcaster once at the beginning.(It may be a bug in ros2) So i suggest you launch your robot at first and listen the tf transform you need at first and launch the realsensel_ros2_camera node. :)

ahuizxc, 2019-11-01:
@MengNan-Li  have you solved this problem?

closed 2019-11-04 by ahuizxc