intel/ros2_intel_realsense #62: exception thrown during user callback

Jconn, 2019-07-30:
When I publish my D435 image data on computer A and subscribe to that data on computer B, I get the following error message in the terminal of the publisher. 

(source.cpp:127) Exception was thrown during user callback!

And running gdb and breaking on caught exceptions yields me this as the source of that error message

librealsense::frame_source::invoke_callback(librealsense::frame_holder) const () from /opt/ros/dashing/lib/librealsense2.so.2

I still get frames from the camera, but I would really like to know why the realsense lib is complaining when I'm on two separate computers.

Jconn, 2019-08-16:
This was being caused by fastRTPS, still haven't figured out the issue but ahs gone away with opensplice

closed 2019-08-16 by Jconn

IanTheEngineer, 2019-09-24:
@Jconn ROS Answers wasn't necessarily a bad place for your question, though I am sorry the community didn't answer it there. If this is indeed an issue purely with Fast RTPS, we should open this as an issue with them. Fast RTPS is the default RMW implementation for ROS2 after all...

Jconn, 2019-09-25:
okay! I will do that