intel/ros2_intel_realsense #67: Experiencing lower frame rate than requested

limjizhengshawn, 2019-08-27:
Hi,

I'm using:
librealsense commit master e5d51792182a71f47fa87e086fa46a9b778a06e8
ros2_intel_realsense commit master 31cdfb7a0f1728fbfbe382e21490231cd69b2978
ROS 2.0 Dashing
Ubuntu 18.04

I have tried both compiling from source and installing through debian.

How to replicate:
Terminal 1: `ros2 run realsense_ros2_camera realsense_ros2_camera`.
Terminal 2: `ros2 topic hz /camera/color/image_raw`.

By default, the driver is supposed to produce images at 30fps.
However, the output at terminal 2 shows frequency lower than that. I've tried in the 3 different protocols:
FastRTPS - ~8-10Hz
OpenSplice - ~15-17Hz
RTI Connext - ~18-20Hz

I'm not sure if this is related to the driver, or an inherent ROS2 problem.

limjizhengshawn, 2019-08-28:
Tested the method suggested by ros2 to transmit images:
`ros2 run image_tools cam2image -b -x 720 -y 1280 -f 30`
`ros2 run image_tools showimage`
`ros2 topic hz /image`

In FastRTPS I'm able to get up to 28 fps.

xhuan28, 2019-09-02:
Have you solved your problems by this way?

limjizhengshawn, 2019-09-06:
@xhuan28 I've tried the refactor branch, and still experience the same issue. Frame rates are significantly lower than the requested 30fps.
FastRTPS - ~8-10Hz
OpenSplice - ~15-17Hz
RTI Connext - ~18-20Hz

I converted the entirety of the v2.2.4 ROS1 driver into ROS2 syntax https://github.com/IntelRealSense/realsense-ros/tree/2.2.4 and was able to get a lot better of a performance:
FastRTPS - ~10-15Hz
RTI Connext - ~29-30Hz

Jconn, 2019-09-26:
Hi @limjizhengshawn, are you willing to share that conversion? I have also been trying to optimize my publish frequencies. 

Also, how did you get the refactor branch building? I have installed librealsense 2.28.1 through apt, and have all my other ros-dashing librealsense packages up to date but am getting build errors.

limjizhengshawn, 2019-10-17:
Hi @Jconn I apologise but I'm unable to share the conversion.

As for the refactor branch building, I installed everything from source instead of relying on installing through apt. I find that doing it via this method would produce lesser errors.