intel/ros2_intel_realsense #89: refactor branch use case

Jconn, 2019-10-24:
I see the refactor branch was added a few weeks ago with some significant changes to the node. Is there some documentation on new features that have been added and how to build the branch?

ahuizxc, 2019-10-25:
Hi Jconn, you can follow the README from refactor branch. In this branch, we support T265 camera and add some launch files for usage :)

> I see the refactor branch was added a few weeks ago with some significant changes to the node. Is there some documentation on new features that have been added and how to build the branch?

Jconn, 2019-10-25:
I get a build error following the readme instructions: 
```
/home/johnconn/external_storage/refactor_ws/src/ros2_intel_realsense/realsense_ros/src/rs_factory.cpp:148:10: error: ‘class rs2::context’ has no member named ‘unload_tracking_module’
     ctx_.unload_tracking_module();
```
I confirmed that my `ros-dashing-librealsense2` is up to date. 
Should I be building librealsense from source for this branch?

ahuizxc, 2019-10-25:
> I get a build error following the readme instructions:
>
> ```
> /home/johnconn/external_storage/refactor_ws/src/ros2_intel_realsense/realsense_ros/src/rs_factory.cpp:148:10: error: ‘class rs2::context’ [...]

update your librealsense may resolve this problem

mfouadthabet, 2019-10-26:
I faced similar problem. 
I found out that installing ros-dashing-librealsense2 as per the instructions caused this problem as it installed an old version on the /opt/ros/dashing/include <== related to v 2.16.0 , which does not include unload_tracking_module.

Removing this package (apt-get purge ros-dashing-librealsense2) and building librealsense latest 2.29.0 from source manage to fix this problem for me.

ahuizxc, 2019-10-28:
> librealsense

this apt package has not been updated for a long time:) . You need to build the librealsense form source.

closed 2019-10-28 by Jconn

portgasray, 2019-11-22:
#93

ManuelMeraz, 2020-04-12:
> > librealsense
> 
> this apt package has not been updated for a long time:) . You need to build the librealsense form source.

For future referense, this is false. I was able to install it with the apt package manager according to the intel realsense documentation for linux and it compiled fine for me:https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md