intel/ros2_intel_realsense #93: No matching function for call to rs2::pipeline::start

portgasray, 2019-11-22:
Error info:
```
/home/lab001/ros2_overlay_ws/src/ros2_intel_realsense/realsense_ros/src/rs_base.cpp: In member function ‘void realsense::RealSenseBase::startPipeline()’:
/home/lab001/ros2_overlay_ws/src/ros2_intel_realsense/realsense_ros/src/rs_base.cpp:67:102: error: no matching function for call to ‘rs2::pipeline::start(rs2::config&, std::_Bind_helper<false, void (realsense::RealSenseBase::*)(const rs2::frame&), realsense::RealSenseBase*, const std::_Placeholder<1>&>::type)’
   pipeline_.start(cfg_, std::bind(&RealSenseBase::publishTopicsCallback, this, std::placeholders::_1));

```

[image]

[image]

[image]

RoboticsYY, 2019-11-22:
@ahuizxc This student came to me for a help, but I'm not sure what is missing here. Could you please help him? Thanks.

portgasray, 2019-11-22:
I think I find the same problem in this issue #89

portgasray, 2019-11-22:
I followed the advice from [mfouadthabet](https://github.com/intel/ros2_intel_realsense/issues/89#issuecomment-546624859), just executed `sudo apt-get purge ros-dashing-librealsense2` and it works for me!