intel/ros2_intel_realsense #102: Eloquent branch: Errors running node

calvertdw, 2020-01-08:
Platform: Jetson Nano 18.04
Sensor: D435

I built a colcon workspace with:

- ros2_intel_realsense (eloquent)
- librealsense (master)

with ROS 2 Eloquent installed via debs.

I put the serial number into the `d435.yml`

Running `ros2 run realsense_node realsense_node __params:=/home/jetson/dev/realsense_ws/src/ros2_intel_realsense/realsense_examples/config/d435.yaml __ns:=/d435`
yields:
```
[WARN] [rcl]: Found parameter file rule '__params:=/home/jetson/dev/realsense_ws/src/ros2_intel_realsense/realsense_examples/config/d435.yaml'. This syntax is deprecated. Use '--ros-args --params-file /home/jetson/dev/realsense_ws/src/ros2_intel_realsense/realsense_examples/config/d435.yaml' instead.
[WARN] [rcl]: Found remap rule '__ns:=/d435'. This syntax is deprecated. Use '--ros-args --remap __ns:=/d435' instead.
[INFO] [d435.camera]: Device with serial number 752112070330 was found.
[INFO] [d435.camera]: Create a node for D4X5 Camera
terminate called after throwing an instance of 'rs2::error'
  what():  Couldn't resolve requests
```

and running `ros2 launch realsense_examples rs_camera.launch.py` yields:
```
[INFO] [launch]: All log files can be found below /home/jetson/.ros/log/2020-01-08-13-00-16-018285-jetson-1614
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense_node-1]: process started with pid [1630]
[realsense_node-1] terminate called after throwing an instance of 'rs2::invalid_value_error'
[realsense_node-1]   what():  hwmon command 0x10 failed. Error type:  (1).
[ERROR] [realsense_node-1]: process has died [pid 1630, exit code -6, cmd '/home/jetson/dev/realsense_ws/install/realsense_node/lib/realsense_node/realsense_node --ros-args'].

```

ahuizxc, 2020-01-10:
what version of librealsense you are using?

> Platform: Jetson Nano 18.04
> Sensor: D435
>
> I built a colcon workspace with:
>
> * ros2_intel_realsense (eloquent)
> * librealsense (master)
>
> with ROS 2 Eloquent installed via debs.
>
> I put the serial [...]

Jconn, 2020-02-14:
This node crashes on startup around 50% of the time for me.
The backtraces on each crash are all over the place, sometimes when discovering the camera, sometimes when receiving a frame. IntelRealSense/librealsense#4158 might be related, but I don't want to spend the cycles figuring out how to replace rs2::pipeline from the node

robotster, 2020-10-04:
@Jconn I am facing the same issue. How did you resolve it?