intel/ros2_intel_realsense #111: node crash on startup

Jconn, 2020-02-14:
This node crashes on startup around 50% of the time for me. 
The backtraces on each crash are all over the place, sometimes when discovering the camera, sometimes when receiving a frame. https://github.com/IntelRealSense/librealsense/issues/4158 might be related, but I don't want to spend the cycles figuring out how to replace `rs2::pipeline` from the node

Jconn, 2020-02-14:
oops, this is a duplicate of https://github.com/intel/ros2_intel_realsense/issues/102

Jconn, 2020-02-17:
I'm reopening this because I'm not sure its a duplicate. My issue is happening with the d435i, and the crash is related to the imu frames. The node will fail to publish any imu data and then crash after 15 seconds with a message about a timeout on an `rs2::frame`

ahuizxc, 2020-02-20:
could you tell me which librealsense version and ros2_intel_realsense you are using?
it may help a lot, thanks!

ahuizxc, 2020-02-20:
> I'm reopening this because I'm not sure its a duplicate. My issue is happening with the d435i, and the crash is related to the imu frames. The node will fail to publish any imu data and then crash after 15 seconds with a message about a timeout on an `rs2::frame`

could you tell me which librealsense version and ros2_intel_realsense you are using?
it may help a lot, thanks!

Jconn, 2020-02-20:
I was using v2.32.1.0 with firmware from the realsense. This was user error on my part. I will close

closed 2020-02-20 by Jconn