intel/ros2_intel_realsense #112: rs2::error when trying to run T265 node

epcava, 2020-02-18:
Device info: 
    Name                          : 	Intel RealSense T265
    Serial Number                 : 	0000908412111187
    Firmware Version              : 	0.2.0.926
    Product Line                  : 	T200

Operating System & Version   Linux (Ubuntu 18.04)
 Kernel Version (Linux Only)       4.15.0-041500-generic 
ROS2                                    Dashing
SDK Version                          2.0 (v2.32.1)
ros2_intel_realsense pkg           2.0.6 (2019-09-03)

I am new to ROS2 and librealsense and I've been trying to get the camera T265 
running with ros2_intel_realsense pkg, however when I try to start a node I get this:

>> ~/ros2_ws$ ros2 run realsense_node realsense_node __params:=`ros2 pkg prefix realsense_examples`/share/realsense_examples/config/t265.yaml __ns:=/t265
[INFO] [t265.camera]: Device's serial number is not set, enabling the default device!
[INFO] [t265.camera]: Device with serial number 0000908412111187 was found.
[INFO] [t265.camera]: Create a node for T265 Camera
terminate called after throwing an instance of 'rs2::error'
  what():  No device connected

The device is being recognised by rs-enumerate-devices and realsense-viewer:

>> ~/ros2_ws$ rs-enumerate-devices 
Device info: 
    Name                          : 	Intel RealSense T265
    Serial Number                 : 	0000908412111187
    Firmware Version              : 	0.2.0.926
    Physical Port                 : 	2-6-3
    Product Id                    : 	0B37
    Usb Type Descriptor           : 	3.1
    Product Line                  : 	T200

Stream Profiles supported by Tracking Module
 Supported modes:
    stream       resolution      fps       format   
    Fisheye 1	  848x800	@ 30Hz	   Y8
    Fisheye 2	  848x800	@ 30Hz	   Y8
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 62Hz	   MOTION_XYZ32F
    Pose	 N/A		@ 200Hz	   6DOF

[image]

Anyone with the same or similar problem?

ahuizxc, 2020-02-20:
> 2.0.6

Hi i suggest you build the project from source code and use refactor branch, thanks:)

zentala, 2020-03-11:
I have the same issue. Building from `refactor` didn't help. Building Real Sense SDK from sources didn't help also. Soon I will provide more details.

ahuizxc, 2020-03-12:
> I have the same issue. Building from `refactor` didn't help. Building Real Sense SDK from sources didn't help also. Soon I will provide more details.

is your realsense-viewer working good?

zentala, 2020-03-13:
Yes, my `realsense-viewer`, `rs-pose`, `rs-capture` and `rs-save-to-disk` are working well. 

I tested on Linux Ubuntu 18.04.4 (kernels `4.15.0-041500-generic`, `4.15.18-041518-generic` and `5.3.0-40-generic`), under 2 different computers (NUC 7i3BNK and Dell XPS 8930). I experimented with different versions of SDKs in range `v2.29.0` - `v2.33.1`. I am using ROS2 Eloquent. I tested `ros2_intel_realsense` from branch `refactor` (and before also `eloquent` too). I didn't work in any case. I am also testing with D435, which is working in every case with ROS2 LRS Node, when T265 doesn't. I tested 3 different USB cables (including one original that came with the camera). I don't use any USB hub. But in any case, I made a few tests with one that is externally powered. Of course, I am always using USB 3.1 ports and cables and I verified it in RealSense Viewer. I also tested those scripts that reboot the computer with 30s delay. Also tried to unplug computer power cable, wait for 30s and plug it again. I executed `cmake` with and without a flag for T265. Without any success. 

Probably this will not be a lead for you but:

Under kernel `5.3.*` I was able to use patch and driver from the same `librealsense v2.33.1` (from the same branch). But it didn't work anyway. But for kernels `4.15.*` I wasn't able to use patch and diver from the same version of SDK. Firstly I tested SKD source from one branch and kernel patch from another. Does it matter? Or should I always use the patch and SDK from the same branch? 

When I wanted to use patch and source from the same branch I got following errors:

Case 1: Dell, kernel `4.15.0-041500-generic`, SDK ` @842ee1e1#v2.33.1` / `@83f952a4#v2.32.1` / `@83f952a4#v2.31.0`, patching with `./scripts/patch-realsense-ubuntu-lts.sh`:

```
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-bionic-master folder

Package required libelf-dev:  - found
Package required elfutils:  - found
Package required bison:  - found
Package required flex:  - found
Fetching Ubuntu LTS tag   to the local kernel sources folder
fatal: You need to specify a tag name.
```

Case 2: Exactly as above but with SDK `@7f841d0d#v2.30.1`:

```
$ ./scripts/patch-realsense-ubuntu-lts.sh                                                                                       ✔  12:55:31  
Reading package lists... Done
Building dependency tree       
Reading state information... Done
bc is already the newest version (1.07.1-2).
build-essential is already the newest version (12.4ubuntu1).
git is already the newest version (1:2.17.1-1ubuntu0.5).
linux-headers-generic is already the newest version (4.15.0.88.80).
The following packages were automatically installed and are no longer required:
  libnvidia-common-415 libnvidia-common-435
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 6 not upgraded.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
The xhci_patch flag is compatible with LTS branch 4.4 only, currently selected kernel is 4.15.0-041500-generic
```
Case 3: Exactly as above but with `@9f99fa9a#v2.30.0` or `@5096aab4#v2.29.0`: `Script has completed.` So I uninstalled RS SKD and installed from sources `@5096aab4#v2.29.0`. So I have kernel patch and SDK from the same branch but then I got an error:

```
$ ros2 run realsense_node realsense_node
/home/zentala/ros2_rs_refactor_ws/install/realsense_node/lib/realsense_node/realsense_node: error while loading shared libraries: librealsense2.so.2.33: cannot open shared object file: No such file or directory
```
I didn't have time to debug it. So here is where I stopped with kernel `4.15`. 

But still... even I was able to use patch and source from `lrs v2.33.1` (the same branch) under kernel 5.3 I still got the same error:
```
$ ros2 run realsense_node realsense_node
[INFO] [camera]: Device's serial number is not set, enabling the default device!
[INFO] [camera]: Device with serial number 944222110877 was found.
[INFO] [camera]: Create a node for T265 Camera
terminate called after throwing an instance of 'rs2::error'
  what():  No device connected
```

Some additional debug info:

```
$ rs-enumerate-devices
Device info: 
    Name                          : 	Intel RealSense T265
    Serial Number                 : 	944222110877
    Firmware Version              : 	0.2.0.926
    Physical Port                 : 	2-1-10
    Product Id                    : 	0B37
    Usb Type Descriptor           : 	3.1
    Product Line                  : 	T200

Stream Profiles supported by Tracking Module
 Supported modes:
    stream       resolution      fps       format   
    Fisheye 1	  848x800	@ 30Hz	   Y8
    Fisheye 2	  848x800	@ 30Hz	   Y8
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 62Hz	   MOTION_XYZ32F
    Pose	 N/A		@ 200Hz	   6DOF

$
```

BTW does this camera supports only `Y8` format?

zentala, 2020-03-16:
I was able to solve this issue:
```
$ ros2 run realsense_node realsense_node
/home/zentala/ros2_rs_refactor_ws/install/realsense_node/lib/realsense_node/realsense_node: error while loading shared libraries: librealsense2.so.2.33: cannot open shared object file: No such file or directory
``` 
by manual removing all RealSense SDK files and installing SDK again. 
The error was caused by the conflict of different SDKs that I installed previously and probably didn't uninstall some of them. 
But still, it didn't solve my problem. 

Any idea of how can I make working mine T265 with ROS2?

zentala, 2020-03-16:
Finally, I made my T265 it working with ROS. 

Stack:
* Ubuntu 18.04.4
* Kernel 5.3 (in my case `5.3.18-050318-generic` but probably 5.3.0 will be easier to patch)
* ROS2 Dashing
* RealSense SDK v2.26.0 built from sources
* `ros2_intel_realsense` from branch `refactor`

Tips:
* In case of errors when building `ros2_intel_realsense` remove `ros-dashing-librealsense2` because it is outdated. Use: `sudo apt purge ros-dashing-librealsense2`. Installing SDK from sources is enough. More: [https://github.com/intel/ros2_intel_realsense/issues/89](https://github.com/intel/ros2_intel_realsense/issues/89)
* According to CHANGELOG current version of `ros2_intel_realsense`supports RealSense SDK `v2.26.0`. Check here to ensure if something changed: [https://github.com/intel/ros2_intel_realsense/blob/refactor/CHANGELOG.rst](https://github.com/intel/ros2_intel_realsense/blob/refactor/CHANGELOG.rst)
* You need to patch kernel. In case of error `fatal: You need to specify a tag name` during patching kernel with `./scripts/patch-realsense-ubuntu-lts.sh`:
  * go to line 92 and add in next lines `echo "$(git ls-remote --tags origin)"`and `echo "${kernel_full_num}"`so you will be able to see available patches and your kernel number
  * your kernel doesn't match to any of those available patches (eg. it's `5.3.18` when supported is only `5.3.0`) and that why this error occurs
  * find closest patch (in my case it was 5.3); edit this script, go to line 92 and replace `kernel_full_num=$(echo $LINUX_BRANCH | cut -d '-' -f 1,2)` with `kernel_full_num=5.3`
  * what should help you to avoid this issue is to use kernel in exact version 5.3.0
* Error: `/usr/bin/ld: warning: libopencv_imgcodecs.so.3.2, needed by /opt/ros/dashing/lib/libcv_bridge.so, may conflict with libopencv_imgcodecs.so.3.4
/usr/bin/ld: warning: libopencv_core.so.3.2, needed by /opt/ros/dashing/lib/libcv_bridge.so, may conflict with libopencv_core.so.3.4` can be easy solved with `sudo apt install ros-dashing-cv-bridge ros-dashing-vision-opencv`

ahuizxc, 2020-03-17:
glad to hear that, there indeed are some bugs related to the OS and we cannot figure it out - -||

zentala, 2020-03-17:
I don't understand. That wasn't an issue of OS, just an issue with patching script and hard to find information that `ros2_intel_realsense` is compatible with `v2.26.0` SDK.

I just set it up under ROS2 Eloquent. The same stack as above, just with Eloquent. It seems that using `v2.26.0` SDK solved this issue. I just had to build a local workspace again. 

It seems that this error:
```
$ ros2 run realsense_node realsense_node
/home/zentala/ros2_rs_refactor_ws/install/realsense_node/lib/realsense_node/realsense_node: error while loading shared libraries: librealsense2.so.2.33: cannot open shared object file: No such file or directory
```
occured when I removed SDK `v2.33.1`. then installed another one but I didn't build workspace again.  
So building local workspace again (`colcon build --symlink-install`) after reinstalling SDK solves this issue. 

I hope you will improve your documentation to provide a working stack. 
It's very time consuming to set up your solutions because of outdated and poor documentation.

ahuizxc, 2020-03-20:
> I don't understand. That wasn't an issue of OS, just an issue with patching script and hard to find information that `ros2_intel_realsense` is compatible with `v2.26.0` SDK.
>
> I just set it up under [...]

sorry for that, man.
Due to adjustment of our company, me and my teammates has moved to another group, so this repo may cannot update in time :( .
so i cannot  guarantee it can function properly if the librealsense is updated.

closed 2020-03-25 by ahuizxc

berndpfrommer, 2020-04-01:
The less painful workaround is to hack the rs_base.cpp as follows:

1) In the constructor, comment the following 2 lines out:
 ```
  //auto sn = dev_.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);                                                                                                                                                        
  //cfg_.enable_device(sn);                                                                                                                                                                                        

```
2) At the very beginning of RealSenseBase::startPipeline(), add the following code:

```
  dev_.hardware_reset();

  if (!cfg_.can_resolve(pipeline_)) {
    throw std::runtime_error("cannot find camera matching config!");
  }

```

keshavchintamani, 2020-04-21:
@berndpfrommer we're having the same issue running realsense ros2 packages at tag 2.0.7 and librealsense2 at 2.34 installed from apt. Tried your workaround but doesnt seem to fix the problem. Could you share some details on your setup?

berndpfrommer, 2020-04-22:
- Ubuntu 18.04
- ROS2 Eloquent
- realsense installed from apt package, Version: 2.34.0-0~realsense0.2251
Did you make the above changes to the "refactor" branch? That's the one for the t265.
https://github.com/intel/ros2_intel_realsense/tree/refactor

keshavchintamani, 2020-04-22:
We are testing at the most recent tag. I will checkout refactor and try it again. Will post the results here. Thanks for the feedback!

keshavchintamani, 2020-04-22:
@berndpfrommer we have the same configuration as you except we have a T265 and a D435i connected to the same host. I forgot to mention this earlier

I get the following reflecting your setup including switching to refactor:

**With a D435i and a T265 connected**

```
ros2 run realsense_node realsense_node --ros-args --params-file `ros2 pkg prefix robot_realsense`/share/robot_realsense/config/t265.yaml --remap __ns:=/t265 __node:=t265
[INFO] [t265.camera]: Device with serial number 90251XXXXXX was found.
[INFO] [t265.camera]: Device with serial number 90841XXXXXX was found.
[INFO] [t265.camera]: Create a node for T265 Camera
terminate called after throwing an instance of 'std::runtime_error'
  what():  cannot find camera matching config!
```
Seems to find two devices ...

**With only T265 connected**

```
ros2 run realsense_node realsense_node --ros-args --params-file `ros2 pkg prefix robot_realsense`/share/robot_realsense/config/t265.yaml --remap __ns:=/t265 __node:=t265
[INFO] [t265.camera]: Device with serial number 908412110938 was found.
[INFO] [t265.camera]: Create a node for T265 Camera
terminate called after throwing an instance of 'std::runtime_error'
  what():  cannot find camera matching config!
```
Finds one serial device this time which is correct and still crashes...

I'm trying to avoid patching the kernel. 

Not sure but as @zentala  mentioned, its a lot of work to get these sensors working....

berndpfrommer, 2020-04-22:
Not sure this makes a difference, but in my launch file, I specify the serial number:
```
t265_serial_no = LaunchConfiguration('serial_no', default="908412110938'')
```
And then later, in the Node() section:
```
        Node(package='realsense_node',
etc etc
       parameters=[{'serial_no':t265_serial_no}]
        )    

```

berndpfrommer, 2020-04-24:
This version works with the t265 on my machine:

https://github.com/berndpfrommer/ros2_intel_realsense/tree/rs_t265_hack

Launch it like this:

ros2 launch realsense_examples rs_t265.launch.py

wilsonthongwk, 2020-10-12:
hi all,

i found a way to work-around the `rs2::error` exception issue.

here is my problem encountered. immediately after running the `realsense_node` on ros2 dashing, it throws out an exception with a `rs2::error` message.

```sh
$ ros2 run realsense_node realsense_node
[INFO] [camera]: Device's serial number is not set, enabling the default device!
[INFO] [camera]: Device with serial number 825312073476 was found.
[INFO] [camera]: Create a node for D4X5 Camera
terminate called after throwing an instance of 'rs2::error'
  what():  Couldn't resolve requests
```

i am working on commit df94e947a2866469612158deb0316eee324e66d2 on branch `dashing`. Following is my work around.

Add one line of code before line 58 in rs_base.cpp (https://github.com/intel/ros2_intel_realsense/blob/dashing/realsense_ros/src/rs_base.cpp#L58) as follow
```diff
  void RealSenseBase::startPipeline()
  {
+   cfg_ = rs2::config();
    auto p_profile = cfg_.resolve(pipeline_);
    auto active_profiles = p_profile.get_streams();
  for (auto & profile : active_profiles) {
```

I am not sure why it works in my PC. maybe the `rs2_create_config()` inside the `rs2::config()` cannot be called too early?

See if this work-around can work for you also.

Thanks,
Wilson