intel/ros2_intel_realsense #113: Multi-camera support.

chuan9, 2020-03-16:
Hi, all
   I use two D435 cameras on ubuntu, and can't launch them successfully.
```
bowen@amp3:~/workspace/depthimage_to_laserscan[(HEAD detached at origin/dashing-devel)*]$ ros2 launch rs_multiple_devices.launch.py
[INFO] [launch]: All log files can be found below /home/bowen/.ros/log/2020-03-16-17-45-44-976443-amp3-13582
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense_node-1]: process started with pid [13599]
[INFO] [realsense_node-2]: process started with pid [13600]
[realsense_node-1] terminate called after throwing an instance of 'rs2::backend_error'
[realsense_node-1]   what():  xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
[ERROR] [realsense_node-1]: process has died [pid 13599, exit code -6, cmd '/opt/robot_devkit/perception/realsense_node/lib/realsense_node/realsense_node __ns:=/camera1 __params:=/tmp/launch_params_wm115vow'].
```

I specify two serial number for cameras in script as bellows:
```
def generate_launch_description():
    # config the serial number and base frame id of each camera
    camera1_base_frame_id = LaunchConfiguration('base_frame_id', default='camera1_link')
    camera2_base_frame_id = LaunchConfiguration('base_frame_id', default='camera2_link')
    camera1_serial_no = LaunchConfiguration('serial_no', default='937422070858')
    camera2_serial_no = LaunchConfiguration('serial_no', default='938422073729')

    camera1_node = Node(
        package='realsense_node',
        node_executable='realsense_node',
        node_namespace="/camera1",
        output='screen',
        parameters=[{'serial_no':camera1_serial_no, 
                    'base_frame_id': camera1_base_frame_id}]
        )
    camera2_node = Node(
        package='realsense_node',
        node_executable='realsense_node',
        node_namespace="/camera2",
        output='screen',
        parameters=[{'serial_no':camera2_serial_no, 
                    'base_frame_id': camera2_base_frame_id}]
        )
    return launch.LaunchDescription([camera1_node, camera2_node])
```

The system config is as bellow:
```
Ubuntu 18.04.3 LTS \n \l
5.3.0-40-generic
realsense-ros: commit e41d491a1cbdb0c4529a651adbc45693db99bd01 (HEAD, origin/refactor)
librealsense version:2.32.1-0~realsense0.1913
```
Will you help have a look at this?

Jconn, 2020-03-16:
I'm hitting this too. The ros2 package can't launch two of the same type of camera. 

If I use a d435 and d435i, the system works

If i use two d435, the second launch d435 will always crash with the above error.

This does not reproduce with realsense-viewer or the head of the ros1 node.

ahuizxc, 2020-03-20:
> I'm hitting this too. The ros2 package can't launch two of the same type of camera.
> 
> If I use a d435 and d435i, the system works
> 
> If i use two d435, the second launch d435 will always crash with the above error.
> 
> This does not reproduce with realsense-viewer or the head of the ros1 node.

Hi jconn,

Sorry for replying you so late, since i am no longer working on this project now.
this problem is already fixed base on the latest PR, please pull the latest code and try it :)

Jconn, 2020-03-20:
@ahuizxc That PR fixed my issue, many thanks! I really appreciate you fixing this while off the project. 

Are there any maintainers on this project right now? @ruffsl has a few PRs that it would be nice to merge.

closed 2020-03-22 by ahuizxc