intel/ros2_intel_realsense #116: Generating point cloud on remote machine

zentala, 2020-03-20:
I would like to connect my D435 to my robot (local machine) and advertise only raw depth & RGB images. Then compute point cloud on the remote machine. Is it possible? How? Any solutions? Suggestions?

ahuizxc, 2020-03-22:
> I would like to connect my D435 to my robot (local machine) and advertise only raw depth & RGB images. Then compute point cloud on the remote machine. Is it possible? How? Any solutions? Suggestions?
> 
> Or maybe... do u have any idea hot to push point cloud via network? I got very slow FPS if I am doing it and it seems like a ROS limitation...
> 
> I am using `ros2_intel_realsense` from branch `refactor` and SDK `v2.26.0`.

the pointcloud message is very large and it may very slow for transform via network. Maybe you can    compute the aligned depth image from remote machine and generate the RGB and aligned depth to pointcloud on your local machine.

zentala, 2020-04-01:
To be honest I have no idea how to do it. Is there any instruction about that?

ahuizxc, 2020-04-02:
> To be honest I have no idea how to do it. Is there any instruction about that?

https://github.com/intel/ros2_intel_realsense/blob/99bbe9197983c31df42ed52cfe6b84b7c41fdb94/realsense_ros2_camera/src/realsense_camera_node.cpp#L1045 this code can generate pointcloud from rgb_img and depth_img , hope it can help you