intel/ros2_intel_realsense #117: infra1 and infra2 topics - where is this images from?

ninamwa, 2020-03-23:
Hi. 
A little (maybe) off-topic question, but I am wondering what images are published to the topics: 
Rectified infra1 image: /camera/infra1/image_rect_raw
Rectified infra2 image: /camera/infra2/image_rect_raw

Is this the images from the Left and Right Imager on the camera (HD image sensors) or is it from the IR projector? 
I guess i am confused because they are named infra, so it sounds like it is an image from the infrared projector, but at the same time, there are two of them? 

I tried to look for documentation of this somewhere, but I could not find any. Thanks

ahuizxc, 2020-03-25:
> HD

Hi nimamwa

These two image topics come from the IR cameras and the depth image come from the IR laser projector，IR camera and IR projector are two different sensors.

Thanks!

ninamwa, 2020-03-25:
Hi. Just to make sure I understand you correctly: the depth image topic is only from the IR laser projector? 

Based on this text (from the D400 series datasheet, p.14) I thought the depth image was a combination from both cameras and the IR projector? 

> The Intel® RealSense™ D400 series depth camera uses stereo vision to calculate
> depth. The stereo vision implementation consists of a left imager, right imager, and an
> optional infrared projector. The infrared projector projects non-visible static IR pattern
> to improve depth accuracy in scenes with low texture. The left and right imagers
> capture the scene and send imager data to the depth imaging (vision) processor,
> which calculates depth values for each pixel in the image by correlating points on the
> left image to the right image and via shift between a point on the Left image and the
> Right image. The depth pixel values are processed to generate a depth frame.
> Subsequent depth frames create a depth video stream.

ahuizxc, 2020-03-25:
you are right :)

closed 2020-04-10 by ahuizxc