intel/ros2_intel_realsense #119: Can't start second camer node

AndreasAZiegler, 2020-04-02:
I'm running ROS2 eloquent on Ubuntu bionic the current ```refactor``` branch.

## Steps to reproduce
Terminal 1:
```
ros2 run realsense_node realsense_node __params:=/path/to/d435i.yaml __ns:=/realsense
```
d435i.yaml
```
/realsense:
  camera:
    ros__parameters:
      serial_no: 845112070204 # d435i
      base_frame_id: d435i_link
      align_depth: true
      enable_pointcloud: false
      dense_pointcloud: false
      color0:
        enabled: true
        resolution: [640,480]
        fps: 30
      depth0:
        enabled: true
        resolution: [640,480]
        fps: 30
      infra1:
        enabled: false
        resolution: [640,480]
        fps: 30
      infra2:
        enabled: false
        resolution: [640,480]
        fps: 30
      accel0:
        enabled: true
      gyro0:
        enabled: true
```
This node runs without problems, I get this output:
```
[WARN] [rcl]: Found remap rule '__ns:=/realsense'. This syntax is deprecated. Use '--ros-args --remap __ns:=/realsense' instead.
[INFO] [realsense.camera]: Device's serial number is not set, enabling the default device!
[INFO] [realsense.camera]: Device with serial number 845112070204 was found.
[INFO] [realsense.camera]: Create a node for D435i Camera
```
Terminal 2:
```
ros2 run realsense_node realsense_node __params:=/path/to/d435.yaml __ns:=/realsense
```
d435.yaml
```
/realsense:
  camera:
    ros__parameters:
      serial_no: 825312073597 # d435
      base_frame_id: d435_link
      align_depth: true
      enable_pointcloud: false
      dense_pointcloud: false
      color0:
        enabled: true
        resolution: [640,480]
        fps: 30
      depth0:
        enabled: true
        resolution: [640,480]
        fps: 30
      infra1:
        enabled: false
        resolution: [640,480]
        fps: 30
      infra2:
        enabled: false
        resolution: [640,480]
        fps: 30
```
This node fails to start. I get this output:
```
[WARN] [rcl]: Found remap rule '__ns:=/realsense'. This syntax is deprecated. Use '--ros-args --remap __ns:=/realsense' instead.
[INFO] [realsense.camera]: Device's serial number is not set, enabling the default device!
[INFO] [realsense.camera]: Device with serial number 845112070204 was found.
[INFO] [realsense.camera]: Create a node for D435i Camera
terminate called after throwing an instance of 'rs2::backend_error'
  what():  xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
```

ahuizxc, 2020-04-02:
> I'm running ROS2 eloquent on Ubuntu bionic the current `refactor` branch.
> 
> ## Steps to reproduce
> Terminal 1:
> 
> ```
> ros2 run realsense_node realsense_node __params:=/path/to/d435i.yaml __ns:=/d435i
> ```
> 
> d435i.yaml
> 
> ```
> /d435i:
>   camera:
>     ros__parameters:
>       serial_no: 845112070204 # d435i
>       base_frame_id: d435i_link
>       align_depth: true
>       enable_pointcloud: false
>       dense_pointcloud: false
>       color0:
>         enabled: true
>         resolution: [640,480]
>         fps: 30
>       depth0:
>         enabled: true
>         resolution: [640,480]
>         fps: 30
>       infra1:
>         enabled: false
>         resolution: [640,480]
>         fps: 30
>       infra2:
>         enabled: false
>         resolution: [640,480]
>         fps: 30
>       accel0:
>         enabled: true
>       gyro0:
>         enabled: true
> ```
> 
> This node runs without problems, I get this output:
> 
> ```
> [WARN] [rcl]: Found parameter file rule '__params:=/home/mtr/ros/safety_field_detector_ws/install/safety-field-intrusion-detector-ros/share/safety-field-intrusion-detector-ros/config/d435i.yaml'. This syntax is deprecated. Use '--ros-args --params-file /home/mtr/ros/safety_field_detector_ws/install/safety-field-intrusion-detector-ros/share/safety-field-intrusion-detector-ros/config/d435i.yaml' instead.
> [WARN] [rcl]: Found remap rule '__ns:=/d435i'. This syntax is deprecated. Use '--ros-args --remap __ns:=/d435i' instead.
> [INFO] [realsense.camera]: Device with serial number 845112070204 was found.
> [INFO] [realsense.camera]: Create a node for D435i Camera
> ```
> 
> Terminal 2:
> 
> ```
> ros2 run realsense_node realsense_node __params:=/path/to/d435.yaml __ns:=/d435
> ```
> 
> d435.yaml
> 
> ```
> /d435:
>   camera:
>     ros__parameters:
>       serial_no: 825312073597 # d435
>       base_frame_id: d435_link
>       align_depth: true
>       enable_pointcloud: false
>       dense_pointcloud: false
>       color0:
>         enabled: true
[... 42 lines cut ...]
> #3  0x00007ffff70ddae6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
> #4  0x00007ffff70ddb21 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
> #5  0x00007ffff70ddd54 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
> #6  0x00007ffff7570300 in rs2::error::handle(rs2_error*) () from /home/mtr/ros/safety_field_detector_ws/install/realsense_ros/lib/librealsense_ros.so
> #7  0x00007ffff7593bdf in rs2::pipeline_profile rs2::pipeline::start<std::_Bind<void (realsense::RealSenseBase::*(realsense::RealSenseBase*, std::_Placeholder<1>))(rs2::frame const&)> >(rs2::config const&, std::_Bind<void (realsense::RealSenseBase::*(realsense::RealSenseBase*, std::_Placeholder<1>))(rs2::frame const&)>) ()
>    from /home/mtr/ros/safety_field_detector_ws/install/realsense_ros/lib/librealsense_ros.so
> #8  0x00007ffff7584c81 in realsense::RealSenseBase::startPipeline() () from /home/mtr/ros/safety_field_detector_ws/install/realsense_ros/lib/librealsense_ros.so
> #9  0x00007ffff756b94b in realsense::RealSenseNodeFactory::startDevice() () from /home/mtr/ros/safety_field_detector_ws/install/realsense_ros/lib/librealsense_ros.so
> #10 0x00007ffff756a7c3 in realsense::RealSenseNodeFactory::init()::{lambda()#1}::operator()() const ()
>    from /home/mtr/ros/safety_field_detector_ws/install/realsense_ros/lib/librealsense_ros.so
> #11 0x00007ffff756ce57 in void std::__invoke_impl<void, realsense::RealSenseNodeFactory::init()::{lambda()#1}>(std::__invoke_other, realsense::RealSenseNodeFactory::init()::{lambda()#1}&&) () from /home/mtr/ros/safety_field_detector_ws/install/realsense_ros/lib/librealsense_ros.so
> #12 0x00007ffff756cae2 in std::__invoke_result<realsense::RealSenseNodeFactory::init()::{lambda()#1}>::type std::__invoke<realsense::RealSenseNodeFactory::init()::{lambda()#1}>(std::__invoke_result&&, (realsense::RealSenseNodeFactory::init()::{lambda()#1}&&)...) ()
>    from /home/mtr/ros/safety_field_detector_ws/install/realsense_ros/lib/librealsense_ros.so
> #13 0x00007ffff756eca8 in decltype (__invoke((_S_declval<0ul>)())) std::thread::_Invoker<std::tuple<realsense::RealSenseNodeFactory::init()::{lambda()#1}> >::_M_invoke<0ul>(std::_Index_tuple<0ul>) () from /home/mtr/ros/safety_field_detector_ws/install/realsense_ros/lib/librealsense_ros.so
> ---Type <return> to continue, or q <return> to quit---
> #14 0x00007ffff756ec64 in std::thread::_Invoker<std::tuple<realsense::RealSenseNodeFactory::init()::{lambda()#1}> >::operator()() ()
>    from /home/mtr/ros/safety_field_detector_ws/install/realsense_ros/lib/librealsense_ros.so
> #15 0x00007ffff756ec34 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<realsense::RealSenseNodeFactory::init()::{lambda()#1}> > >::_M_run() ()
>    from /home/mtr/ros/safety_field_detector_ws/install/realsense_ros/lib/librealsense_ros.so
> #16 0x00007ffff71086df in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
> #17 0x00007ffff63e76db in start_thread (arg=0x7fffd11d9700) at pthread_create.c:463
> #18 0x00007ffff6b6388f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
> ```

Have you tried on the latest refactor branch? this problem just fixed recently.

AndreasAZiegler, 2020-04-02:
Thanks for the hint @ahuizxc. Indeed, I pulled the ```refactor``` branch from the wrong repository and was out of date.

closed 2020-04-02 by AndreasAZiegler