intel/ros2_intel_realsense #120: Can't start more than one camera node as ComposableNode

AndreasAZiegler, 2020-04-02:
After being able to start two cameras as nodes (#119) I want to start the two camera nodes from a launch file as ```ComposableNode``` to allow inter-process communication.

Running the following launch file
```
import launch
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode
from launch.substitutions import LaunchConfiguration
from ament_index_python import get_package_share_directory

def generate_launch_description():
    """Generate launch description with multiple components."""

    # config the serial number and base frame id of each camera
    d435_base_frame_id = LaunchConfiguration('base_frame_id', default='camera1_link')

    d435_serial_no = LaunchConfiguration('serial_no', default='837212070294')

    container = ComposableNodeContainer(
            node_name='safety_field_intrusion_detection',
            node_namespace='',
            package='rclcpp_components',
            node_executable='component_container',
            composable_node_descriptions=[
                ComposableNode(
                    package='realsense_ros',
                    node_plugin='realsense::RealSenseNodeFactory',
                    node_name='D435',
                    node_namespace='/realsense',
                    parameters=[
                        get_package_share_directory('safety-field-intrusion-detector-ros')+"/config/d435.yaml"
                ]),
                ComposableNode(
                    package='realsense_ros',
                    node_plugin='realsense::RealSenseNodeFactory',
                    node_name='D435i',
                    node_namespace='/realsense',
                    parameters=[
                        get_package_share_directory('safety-field-intrusion-detector-ros')+"/config/d435i.yaml"
                ]),
                ComposableNode(
                    package='safety-field-intrusion-detector-ros',
                    node_plugin='SafetyFieldIntrusionDetectorBaseNode',
                    node_name='safety_field_intrusion_detector',
                    parameters=[
                        get_package_share_directory('safety-field-intrusion-detector-ros')+"/config/params.yaml"
                ])
            ],
            output='screen',
    )

    return launch.LaunchDescription([container])
```
gives this output
```
[INFO] [launch]: All log files can be found below /home/mtr/.ros/log/2020-04-02-10-34-10-493702-UP2-board-17377
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [17388]
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/realsense/D435' in container '/safety_field_intrusion_detection'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/realsense/D435i' in container '/safety_field_intrusion_detection'
[component_container-1] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/safety_field_intrusion_detection' in container '/safety_field_intrusion_detection'
[component_container-1] terminate called after throwing an instance of 'rs2::backend_error'
[component_container-1]   what():  xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
[ERROR] [component_container-1]: process has died [pid 17388, exit code -6, cmd '/opt/ros/eloquent/lib/rclcpp_components/component_container --ros-args -r __node:=safety_field_intrusion_detection'].
```
or
```
[INFO] [launch]: All log files can be found below /home/mtr/.ros/log/2020-04-02-10-38-21-156610-UP2-board-17492
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [17512]
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/realsense/D435' in container '/safety_field_intrusion_detection'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/realsense/D435i' in container '/safety_field_intrusion_detection'
[component_container-1] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/safety_field_intrusion_detection' in container '/safety_field_intrusion_detection'
[component_container-1] terminate called after throwing an instance of 'rs2::invalid_value_error'
[component_container-1]   what():  hwmon command 0x30 failed.
[component_container-1] Error type: No data to return (-21).
[ERROR] [component_container-1]: process has died [pid 17512, exit code -6, cmd '/opt/ros/eloquent/lib/rclcpp_components/component_container --ros-args -r __node:=safety_field_intrusion_detection']
```

AndreasAZiegler, 2020-04-03:
I figured out, that the problem comes from the fact that I want to use the config parameters from the yaml file. When I pass the parameters as variables, it does work.

I will further investigate as I would like to be able to use my config files with ```ros2 lauch...``` as well as with ```ros2 run...```

AndreasAZiegler, 2020-04-06:
I added my configuration files in the first post. As these configuration files do work when I start the nodes with ```ros2 run`` but don't when I use the launch file, I assume there is a bug in the way the config is loaded when using ```ComposableNode```.

ahuizxc, 2020-04-10:
> I added my configuration files in the first post. As these configuration files do work when I start the nodes with `ros2 run` but don't when I use the launch file, I assume there is a bug in the way the config is loaded when using `ComposableNode`.

Hi 
The composableNode API may have some bugs and we have already tried on. Then we gived up and decided to launch nodes separately for multi-camera :) thanks

AndreasAZiegler, 2020-04-17:
@ahuizxc But this way it's not possible to have inter-process communication. This is especially helpful for data types which are data intensive.

mrlowalowa, 2020-04-30:
@AndreasAZiegler the latest FastRTSP Release support shared memory communication. This should help you with the higher data load.
https://discourse.ros.org/t/ros2-default-behavior-wifi/13460/31

note: you need to get the latest FastRTSP version and build it from source in order to use it.

AndreasAZiegler, 2020-04-30:
@mrlowalowa Thanks for the hint. As we have a native rti DDS system with which I might have to communicate, I currently don't consider FastRTSP unless ROS2 supports multiple rmw's.