intel/ros2_intel_realsense #121: realsense_ros exception when launching T265

keshavchintamani, 2020-04-10:
We've been struggling with an exception thrown trying to start our T265 camera

Issue: Whats really wierd is our D435i starts fine but when the T265 starts, we get this exception:

```
$ ros2 run realsense_node realsense_node --ros-args -p serial_no:=908412110938
[INFO] [camera]: Device with serial number 908412110938 was found.
[INFO] [camera]: Create a node for T265 Camera
terminate called after throwing an instance of 'rs2::error'
  what():  No device connected

```
**Our setup:**
ros2 eloquent from repos
ros2_intel_realsense : branch eloquent
librealsense2 from repos at 2.34.0

**Troubleshooting**
Trouble persists with branch refactor
Same Node is able to start a D435i 
realsense-viewer/rs-enumerate-devices is able to stream/enumerate device

Copy of stack trace

```
1  raise                                                                                                                                                                  0x7ffff3c4be97 
2  abort                                                                                                                                                                  0x7ffff3c4d801 
3  ??                                                                                                                                                                     0x7ffff4640957 
4  ??                                                                                                                                                                     0x7ffff4646ae6 
5  std::terminate()                                                                                                                                                       0x7ffff4646b21 
6  __cxa_throw                                                                                                                                                            0x7ffff4646d54 
7  rs2::error::handle                                                                                                                                 rs_types.hpp    156 0x555555582bee 
8  rs2::config::resolve                                                                                                                               rs_pipeline.hpp 321 0x55555559c529 
9  realsense::RealSenseBase::startPipeline                                                                                                            rs_base.cpp     58  0x55555558fb7b 
10 realsense::RealSenseNodeFactory::startDevice                                                                                                       rs_factory.cpp  99  0x55555557e36a 
11 realsense::RealSenseNodeFactory::<lambda()>::operator()(void) const                                                                                rs_factory.cpp  61  0x55555557d1c7 
12 std::__invoke_impl<void, realsense::RealSenseNodeFactory::init()::<lambda()>>(std::__invoke_other, realsense::RealSenseNodeFactory::<lambda()> &&) invoke.h        60  0x55555557f877 
13 std::__invoke<realsense::RealSenseNodeFactory::init()::<lambda()>>(realsense::RealSenseNodeFactory::<lambda()> &&)                                 invoke.h        95  0x55555557f502 
14 std::thread::_Invoker<std::tuple<realsense::RealSenseNodeFactory::init()::<lambda()>>>::_M_invoke<0>(std::_Index_tuple<0>)                         thread          234 0x5555555816c8 
15 std::thread::_Invoker<std::tuple<realsense::RealSenseNodeFactory::init()::<lambda()>>>::operator()(void)                                           thread          243 0x555555581684 
16 std::thread::_State_impl<std::thread::_Invoker<std::tuple<realsense::RealSenseNodeFactory::init()::<lambda()>>>>::_M_run(void)                     thread          186 0x555555581654 
17 ??                                                                                                                                                                     0x7ffff46716df 
18 start_thread                                                                                                                                                           0x7ffff4d7f6db 
19 clone                                                                                                                                                                  0x7ffff3d2e88f 
```
Issue seems to be here:

 ```
pipeline_profile resolve(std::shared_ptr<rs2_pipeline> p) const
        {
            rs2_error* e = nullptr;
            auto profile = std::shared_ptr<rs2_pipeline_profile>(
                rs2_config_resolve(_config.get(), p.get(), &e),
                rs2_delete_pipeline_profile);

            error::handle(e);
            return pipeline_profile(profile);
        }
```

SomaGallai, 2020-05-01:
I got the same problem, did you figure out the problem?

ManuelMeraz, 2020-05-01:
I actually did some research on this @SomaGallai because I experienced this about 2 weeks ago. I'm certain the issue is to do with an older version of librealsense, specifically the ros2 nodes using that older version. 

The issue arised when the D435 is connected at the same time with the T265; also the order in which you plug them in I believe. The latest release of these nodes using v2.2.6 of lib realsense. The right way to use this would be to not use the dashing librealsense, but to build v2.26 from source and run that software to emulate this. It will have trouble detecting both cameras sometimes. 

You can see a similar issue here in the librealsense issues https://github.com/IntelRealSense/librealsense/issues/5509

The correction to this would be for the ros2 nodes to support the latest version of librealsense, because i think this would fix these pipeline issues.

I'll probably contribute this if it isn't done soon because I'm at the point in my personal project where I want to start using my realsense cameras and have everything else mostly working well.

keshavchintamani, 2020-05-01:
So most of my tests were on a Jetson Nano. Althought I must say I was running a L4T 32.3.1 which is quite old. I did realize this could have caused issues due to missing kernel modules etc. 

However I've tried what @ManuelMeraz  suggested on an x86 machine with Ubuntu 18.04. I have the same problem there. 

I've tried combinations from V2.26 until V2.34 from both source and binary and I get the same unstable behavior. 

What makes it difficult is the two tools:
rs-enumerate-devices - > sometimes shows T265 and D435i connected and sometimes only T265
rs-pose - always returns pose from a connected T265. 
realsense-viewer never allows me to see both devices connected at the same time.

Not sure what to make of this. I'm in the process of switching to V2.26 and testing again. 

@ManuelMeraz @SomaGallai  if you have any luck, please share here

ManuelMeraz, 2020-05-01:
@keshavchintamani when you installed the latest lib realsense did you purge all old libraries?

from https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md:
Important Removing Debian package is allowed only when no other installed packages directly refer to it. For example removing librealsense2-udev-rules requires librealsense2 to be removed first.

Remove a single package with:
`sudo apt-get purge <package-name>`

Remove all RealSense™ SDK-related packages with:
`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

In particular the last line. When I've tried the newest realsense I had no issues with this anymore.

keshavchintamani, 2020-05-01:
I remembee using apt remove librealsense* but didn't purge. But I did check
via

$ ldconfig -p | grep libreal

But I will document all settings and retry.

keshavchintamani, 2020-05-04:
I decided to re-run some tests on a clean Jetson Nano which is my target machine 

**Configuration:**
L4T 32.4.2
Kernel 4.9.140
ROS eloquent
Librealsense from source - Tried V2.31.0
Build with flags   -DBUILD_EXAMPLES=true -DFORCE_LIBUVC=true -DBUILD_WITH_CUDA="$USE_CUDA" DBUILD_PYTHON_BINDINGS=bool:true

**With LIBUVC on V2.31.0**
rs-enumerate-devices sees both devices
realsense-viewer captures both device streams
D435i node works ok in standalone mode 
T265 node does not find device in standalone mode 

```
$ ros2 run realsense_node realsense_node --ros-args -p serial_no:=908412110999

[ERROR] [camera]: The Device with serial number 908412110999 is not found. Please make sure it is connected.
[INFO] [camera]: Device with serial number 902512070888 was found.
[INFO] [camera]: Device with serial number 0000908412110999 was found.
```
With D435i node running, T265 node crashes with: 
```
$ ros2 run realsense_node realsense_node --ros-args -p serial_no:=908412110999
terminate called after throwing an instance of 'rs2::error'
  what():  failed to set power state
```

**With LIBUSB (RS_V4L_BACKEND) on V2.31.0**
rs-enumerate-devices only sees T265
realsense-viewer captures both device streams
D435i node does not  ok in standalone mode 
T265 node does not find device in standalone mode 

```
$ ros2 run realsense_node realsense_node --ros-args -p serial_no:=902512070888
[ERROR] [camera]: No RealSense devices were found!
```
That may solve a first hurdle. Seems libuvc is a required backend for the D435i? 

Besides this, I still cant figure out why T265 fails when the D435i is connected, besides what I reported here previously. 

If anyone could replicate this on a Ubuntu 18.04 machine, could shed some light on the problem...

keshavchintamani, 2020-05-04:
Some success now with V2.31.0 (see above) and compiled with LIBUVC.

First launch t265
`$ ros2 run realsense_node realsense_node --ros-args -p serial_no:=908412110999 -p initial_reset:=true -r __ns:=/t265 `

Next launch d435 
`$ ros2 run realsense_nodrealsense_node --ros-args -p serial_no:=902512070888 -r __ns:=/d435`

For the D435i I get this:
`23:09:47.933 [11256] [E] Device-B2E0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)`

but it automatically supresses after a few messages. 

After that 
```
$ ros2 topic list 
/d435/camera/accel/imu_info
/d435/camera/accel/sample
/d435/camera/aligned_depth_to_color/camera_info
/d435/camera/aligned_depth_to_color/image_raw
/d435/camera/color/camera_info
/d435/camera/color/image_raw
/d435/camera/depth/camera_info
/d435/camera/depth/image_rect_raw
/d435/camera/gyro/imu_info
/d435/camera/gyro/sample
/d435/camera/infra1/camera_info
/d435/camera/infra1/image_rect_raw
/d435/camera/infra2/camera_info
/d435/camera/infra2/image_rect_raw
/d435/camera/pointcloud
/d435/parameter_events
/d435/rosout
/parameter_events
/rosout
/t265/camera/accel/imu_info
/t265/camera/accel/sample
/t265/camera/fisheye1/camera_info
/t265/camera/fisheye1/image_raw
/t265/camera/fisheye2/camera_info
/t265/camera/fisheye2/image_raw
/t265/camera/gyro/imu_info
/t265/camera/gyro/sample
/t265/camera/odom/sample
/t265/parameter_events
/t265/rosout
/tf_static
```
Not the best of outcomes, but surely a lot better than nothing... Hope it works for you.

SomaGallai, 2020-05-05:
I'm running on a Jetson TX2, Ubuntu 18.04, 4.9.140-tegra kernel and ROS2 Eloquent. I purged all librealsense and installed 2.26.0 from source now I can run it with a T265. Using this command: `ros2 run realsense_node realsense_node --ros-args -p serial_no:=943222110295 -p initial_reset:=true -r __ns:=/t265` . I am not planning on using an other camera at the same time so it is perfect for me. Thanks!

ManuelMeraz, 2020-05-06:
Can confirm! Realsense V2.31.0 works with both cameras and no issues minus the 
`23:09:47.933 [11256] [E] Device-B2E0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)` error

I'm using a launch file to launch my t265 and d435 and loading them from a single configuration file.

akirayou, 2020-05-09:
FYI:
I can run T265(pose)+D435i(point cloud/depth) with  partial workaround for  librealsense2.34.0  and refactor branch of ros2_intel_realsense.  

Here is the commit of my workaround branch. 
https://github.com/akirayou/ros2_intel_realsense/commit/6d9c8097e213994d27f67fbff3cd7cc6d3a7dbfd

T265 can not do enable_device() which specify device with serial number. So I just skip it on T265.
So you can not use 2 more T265 device. (D435 cameras are ok.)
You can select correct pose or fisheye stream on a T265,because there is only one stream of POSE or FISHEYE.  But  sometimes you can not select correct Gyro or Acc (D435i also have Gyro and Acc).

knatsuki, 2020-05-15:
There seems to be two or so (possibly related) bugs causing problem with t265 device.
1. t265 device cannot be enabled via serial number.  The device query done with pipeline resolve appears to be filtering out non-uvc usb devices, which is preventing t265 device from being recognized. (https://github.com/IntelRealSense/librealsense/issues/5614) 
2. It may be specific to my system, but additional problems are causing runtime error when I try to start the t265 device. Running hardware reset at the start of the program seems to fix those issues. Weirdly, running the same hardware reset on the d435 causes it to disconnect and prevents it from starting 🤨...

It's all monkey patch but my workaround can be found in https://github.com/intel/ros2_intel_realsense/compare/refactor...knatsuki:bug_121. It's essentially identical to @akirayou 's solution.