intel/ros2_intel_realsense #125: Change prefix of camera optical and depth frames in Eloquent branch

charlotteheggem, 2020-05-07:
Hi,
I can not find any parameter to change the prefix for the frames on the Eloquent branch. When launching with two D435 cameras, both camera links, defined with base_frame_id, have the child frames camera_color_optical_frame and camera_depth_optical_frame, and it is not possible to separate these. Is there any way to change the prefix of the frames?

akirayou, 2020-05-15:
Here is my patch. Not tested well, but it seems to work.
All optical frames name  are generated by base_frame_name.

patch:
https://github.com/akirayou/ros2_intel_realsense/commit/7238ceaff0fed56057702573fba2fa95d20682a4

My branch for test:
https://github.com/akirayou/ros2_intel_realsense/tree/T265_workaround

AndrejOrsula, 2020-05-18:
Hey,
my approach, https://github.com/AndrejOrsula/ros2_intel_realsense/commit/6e1365f0db167a6fd190d8ee23542eacba115b16,  was to prefix all frames with the namespace of the node (besides base_frame that remains defined by `base_frame_id`). It is tested with combination of D415 + D435.

That being said, I am really curious to know the standard way of remapping frame_id in ROS 2, as I cannot find any relevant information about it.