intel/ros2_intel_realsense #127: Starting a D435 and a D435i camera fails

AndreasAZiegler, 2020-05-11:
|||
|-|-|
| OS | Ubuntu Bionic |
| Hardware | UP2 board |
| ROS2 version | eloquent |
| ros2_intel_realsense branch | refactor |

I try to start two realsense nodes and another node. When I start them as composable nodes with:
```
import launch
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode
from launch.substitutions import LaunchConfiguration
from ament_index_python import get_package_share_directory

def generate_launch_description():
    """Generate launch description with multiple components."""

    # config the serial number and base frame id of each camera
    d435_base_frame_id = LaunchConfiguration('base_frame_id', default='d435_link')
    d435_serial_no = LaunchConfiguration('serial_no', default='825312073597')
    
    d435i_base_frame_id = LaunchConfiguration('base_frame_id', default='d435i_link')
    d435i_serial_no = LaunchConfiguration('serial_no', default='841512070768')

    container = ComposableNodeContainer(
            node_name='safety_field_intrusion_detection',
            node_namespace='',
            package='rclcpp_components',
            node_executable='component_container',
            composable_node_descriptions=[
                ComposableNode(
                    package='realsense_ros',
                    node_plugin='realsense::RealSenseNodeFactory',
                    node_name='D435',
                    node_namespace='/realsense',
                    parameters=[
                        {'serial_no': d435_serial_no,
                         'base_frame_id': d435_base_frame_id}
                    ]),
                ComposableNode(
                    package='realsense_ros',
                    node_plugin='realsense::RealSenseNodeFactory',
                    node_name='D435i',
                    node_namespace='/realsense',
                    parameters=[
                        {'serial_no': d435i_serial_no,
                         'base_frame_id': d435i_base_frame_id}
                    ]),
                ComposableNode(
                    package='safety-field-intrusion-detector-ros',
                    node_plugin='DetectorRobotComponent',
                    node_name='safety_field_intrusion_detector',
                    parameters=[
                        get_package_share_directory('safety-field-intrusion-detector-ros')+"/config/params.yaml"
                    ])
            ],
            output='screen',
    )

    return launch.LaunchDescription([container])
```
I get the following output:
```
[INFO] [launch]: All log files can be found below /home/mtr/.ros/log/2020-05-11-09-13-46-695913-mtr-UP-APL01-7685
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [7696]
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/realsense/D435' in container '/safety_field_intrusion_detection'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/realsense/D435i' in container '/safety_field_intrusion_detection'
[component_container-1] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/safety_field_intrusion_detection' in container '/safety_field_intrusion_detection'
[component_container-1] terminate called after throwing an instance of 'rs2::error'
[component_container-1]   what():  Couldn't resolve requests
[ERROR] [component_container-1]: process has died [pid 7696, exit code -6, cmd '/opt/ros/eloquent/lib/rclcpp_components/component_container --ros-args -r __node:=safety_field_intrusion_detection'].
```

When I start them with:
```
import os
import launch
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch_ros.actions import Node
from launch.substitutions import LaunchConfiguration
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import ThisLaunchFileDir
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

def generate_launch_description():

    # config the serial number and base frame id of each camera
    d435_base_frame_id = LaunchConfiguration('base_frame_id', default='d435_link')
    d435_serial_no = LaunchConfiguration('serial_no', default='825312073597')
    
    d435i_base_frame_id = LaunchConfiguration('base_frame_id', default='d435i_link')
    d435i_serial_no = LaunchConfiguration('serial_no', default='841512070768')

    d435_node = Node(
        package='realsense_node',
        node_namespace='/realsense',
        node_executable='realsense_node',
        output='screen',
        parameters=[
            {'serial_no': d435_serial_no,
             'base_frame_id': d435_base_frame_id}
            #get_package_share_directory('safety-field-intrusion-detector-ros')+"/config/d435.yaml"
        ])
    d435i_node = Node(
        package='realsense_node',
        node_namespace='/realsense',
        node_executable='realsense_node',
        output='screen',
        parameters=[
            {'serial_no': d435i_serial_no,
             'base_frame_id': d435i_base_frame_id}
            #get_package_share_directory('safety-field-intrusion-detector-ros')+"/config/d435i.yaml"
        ])
    detector_node = Node(
        package='safety-field-intrusion-detector-ros',
        node_executable='mt-detector-robot-node',
        node_name='safety_field_intrusion_detector',
        parameters=[
            get_package_share_directory('safety-field-intrusion-detector-ros')+"/config/params.yaml"
        ])

    return launch.LaunchDescription([d435_node, d435i_node, detector_node])
```
I get the following output:
```
[INFO] [launch]: All log files can be found below /home/mtr/.ros/log/2020-05-11-09-13-11-986002-mtr-UP-APL01-5750
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense_node-1]: process started with pid [5761]
[INFO] [realsense_node-2]: process started with pid [5762]
[INFO] [mt-detector-robot-node-3]: process started with pid [5763]
[realsense_node-1] terminate called after throwing an instance of 'rs2::error'
[realsense_node-1]   what():  Couldn't resolve requests
[ERROR] [realsense_node-1]: process has died [pid 5761, exit code -6, cmd '/home/mtr/ros/safety_field_detector_ws/install/realsense_node/lib/realsense_node/realsense_node --ros-args --params-file /tmp/launch_params_vvxtb6v1 -r __ns:=/realsense'].
```

Only if I start them from a bash script with:
```
ros2 run realsense_node realsense_node __params:=`ros2 pkg prefix safety-field-intrusion-detector-ros`/share/safety-field-intrusion-detector-ros/config/d435i.yaml __ns:=/realsense & \
ros2 run realsense_node realsense_node __params:=`ros2 pkg prefix safety-field-intrusion-detector-ros`/share/safety-field-intrusion-detector-ros/config/d435.yaml __ns:=/realsense & \
ros2 run safety-field-intrusion-detector-ros mt-detector-robot-node
```
the nodes start successfully.
```
[WARN] [rcl]: Found parameter file rule '__params:=/home/mtr/ros/safety_field_detector_ws/install/safety-field-intrusion-detector-ros/share/safety-field-intrusion-detector-ros/config/d435.yaml'. This syntax is deprecated. Use '--ros-args --params-file /home/mtr/ros/safety_field_detector_ws/install/safety-field-intrusion-detector-ros/share/safety-field-intrusion-detector-ros/config/d435.yaml' instead.
[WARN] [rcl]: Found remap rule '__ns:=/realsense'. This syntax is deprecated. Use '--ros-args --remap __ns:=/realsense' instead.
[WARN] [rcl]: Found parameter file rule '__params:=/home/mtr/ros/safety_field_detector_ws/install/safety-field-intrusion-detector-ros/share/safety-field-intrusion-detector-ros/config/d435i.yaml'. This syntax is deprecated. Use '--ros-args --params-file /home/mtr/ros/safety_field_detector_ws/install/safety-field-intrusion-detector-ros/share/safety-field-intrusion-detector-ros/config/d435i.yaml' instead.
[WARN] [rcl]: Found remap rule '__ns:=/realsense'. This syntax is deprecated. Use '--ros-args --remap __ns:=/realsense' instead.
[INFO] [realsense.camera]: Device with serial number 825312073597 was found.
[INFO] [realsense.camera]: Device with serial number 825312073597 was found.
[INFO] [realsense.camera]: Create a node for D4X5 Camera
[INFO] [realsense.camera]: Device with serial number 841512070768 was found.
[INFO] [realsense.camera]: Create a node for D435i Camera
```

akirayou, 2020-05-15:
>    d435_serial_no = LaunchConfiguration('serial_no', default='825312073597')
 >    d435i_serial_no = LaunchConfiguration('serial_no', default='841512070768')

You set same paramter name  for d435 and d435i
May be you want to do this? 

d435_serial_no = LaunchConfiguration('d435_serial_no', default='825312073597')
d435i_serial_no = LaunchConfiguration('d435i_serial_no', default='841512070768')

Here is my example:

    t265_serial_no = LaunchConfiguration('t265_serial_no', default='852212110070')
    d435_serial_no = LaunchConfiguration('d435_serial_no', default='848312071481')
    t265_conf=get_package_share_directory('robo')+'/config/t265.yaml'
    d435i_conf=get_package_share_directory('robo')+'/config/d435i.yaml'
    n=[]
    n.append(Node(
        package='realsense_node',
        node_executable='realsense_node',
        node_namespace="/t265",
        output='screen',
        parameters=[t265_conf,{'serial_no':t265_serial_no, 
                    'base_frame_id': 't265_link'}],  #This means partial overwrite the yaml config of "serial_no" andbase_frame_id
        #arguments=[ '__log_level:=debug'],
        ))
    n.append(Node(
        package='realsense_node',
        node_executable='realsense_node',
        node_namespace="/d435",
        output='screen',
        parameters=[d435i_conf,{'serial_no':d435_serial_no, 
                    'base_frame_id': 'd435_link'}],
              #  arguments=[('__log_level:=debug')],
        ))

AndreasAZiegler, 2020-05-18:
@akirayou Thanks for your input. This indeed solves the issue. I was not aware that having the same parameter name within ```LaunchConfiguration()``` is a problem. Good to know.

closed 2020-05-18 by AndreasAZiegler