intel/ros2_intel_realsense #134: ROS2 Foxy Install

NathanDodson, 2020-06-23:
I'm trying to follow the steps in the read.md file at https://github.com/intel/ros2_intel_realsense/tree/refactor to install. I run sudo apt-get install ros-foxy-librealsense2, but the package does not exist. Anyone else get this issue?

bobgates, 2020-06-24:
What are you building on? IIRC apt only works on Intel Linux, not things like the Raspberry Pi. Here's the bit on the website:

Intel® RealSense™ SDK 2.0 provides installation packages for Intel X86/AMD64-based Debian distributions in dpkg format for Ubuntu 16/18 LTS. (on https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)

That page also gives some instructions on how to add the repos.

Building from sources isn't too bad if you have to go that way.

NathanDodson, 2020-06-24:
I'm trying to build on my laptop. I'm not able to use the dpkg format since Foxy is only supported in Ubuntu 20 LTS. The link I provided refers to Foxy installs, but the command doesn't exist. I confirmed this with another person as well. @bobgates

caelinsutch, 2020-07-01:
Can confirm this is also an issue

caelinsutch, 2020-07-02:
Even after building from source, it doesn't detect librealsense2 as installed :(

dschori, 2020-07-17:
Same issue here, any solution to this?

ROS: Foxy (installed via Debian Packages)

sharronliu, 2020-07-28:
The Foxy release has been submitted to rosditro, and now waiting for verification to get into the Foxy build. Thanks!

surfertas, 2020-08-19:
Confirmed that I was able to install by running `sudo apt-get install ros-foxy-librealsense2` and successfully work through the instructions provided on the refactor branch.
`

mkeyno, 2020-11-04:
@surfertas I followed your approach but received the following error 
any hint?

```
ros@ubuntu:~$ sudo apt-get install ros-foxy-librealsense2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-foxy-librealsense2 is already the newest version (2.34.0-3focal.20201006.191738).
The following packages were automatically installed and are no longer required:
  libfprint-2-tod1 libllvm9 linux-image-5.4.0-26-generic
  linux-modules-5.4.0-26-generic linux-modules-extra-5.4.0-26-generic
  python3-gpg python3-ldb python3-tdb realsense-sdk-udev-rules
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 12 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] 
Setting up librealsense2-dkms (1.3.12-0ubuntu1) ...
Removing old librealsense2-dkms-1.3.12 DKMS files...

------------------------------
Deleting module version: 1.3.12
completely from the DKMS tree.
------------------------------
Done.
Loading new librealsense2-dkms-1.3.12 DKMS files...
Building for 5.4.0-52-generic
Building initial module for 5.4.0-52-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/librealsense2-dkms.0.crash'
Error! Bad return status for module build on kernel: 5.4.0-52-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.12/build/make.log for more information.
dpkg: error processing package librealsense2-dkms (--configure):
 installed librealsense2-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 librealsense2-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

```

cardboardcode, 2021-03-07:
> Confirmed that I was able to install by running `sudo apt-get install ros-foxy-librealsense2` and successfully work through the instructions provided on the refactor branch.
> `
> This was done on:
> Ubuntu: 20.04.1 LTS
> Kernel: 5.4.0-42-generic
> Device: D435 firmware version: 05.12.06.00
> Note: The docs indicate kernel 5.4* and Ubuntu 20.04 arent officially supported as of some unspecified date in the past. I am not sure what is the up to date case now. Maybe @sharronliu can clarify.
> 
> I still run into run time issues though...will open a separate issue for this.

Verified that using the code and instructions under the **refactor** branch works in **manually** setting up the ros2_intel_realsense package in ROS2 Foxy successfully.

## Instructions
The section below list the steps taken in setting up for use:
```bash
# Set up ROS2 workspace
cd $HOME
mkdir -p ~/realsense_ws/src && cd ~/realsense_ws/src

# Download refactor branch of ros2_intel_realsense
git clone https://github.com/intel/ros2_intel_realsense --branch refactor --depth 1

# Build package
cd ~/realsense_ws
source /opt/ros/foxy/setup.bash
colcon build

# Connect camera to workstation.

# Run package
source install/setup.bash
ros2 run realsense_node realsense_node
```