intel/ros2_intel_realsense #141: T265 Node will not run

Michael-Equi, 2020-07-15:
OS: Ubuntu 20.04
ROS dist: Foxy Binaries
Librealsense2 version: 2.36.0-0~realsense0.3169

Running `ros2 run realsense_node realsense_node` with a T265 plugged (no other cameras) causes the following error every time
```

[INFO] [1594789151.355221427] [camera]: Device's serial number is not set, enabling the default device!
[INFO] [1594789151.364391450] [camera]: Device with serial number 948422110355 was found.
[INFO] [1594789151.364452588] [camera]: Create a node for T265 Camera
2020-07-14 21:59:24,588 WARNING [default] Could not open device Unable to open device interface
2020-07-14 21:59:24,591 WARNING [default] Could not open device Unable to open device interface
2020-07-14 21:59:24,594 WARNING [default] Could not open device Unable to open device interface
terminate called after throwing an instance of 'rs2::error'
  what():  No device connected
Signal: SIGABRT (Aborted)
```

EgOrlukha, 2020-07-16:
I have the same issue as @Michael-Equi.
Ubuntu 18.04, ROS2 Dashing, librealsense 2.36

robotsoft, 2020-07-24:
I have the same issue wtih Ubuntu 20.04, ROS2 foxy, librealsense 2.36. Is there any solution?

Michael-Equi, 2020-07-25:
Its a bit raw but I am working on a custom T200 series only ROS2 node for Foxy and 20.04. Works more reliably, uses the lifecycle node interface, and utilizes a few more features of the T265 (map saving/load and relocalization as well as wheel odometry input). Lots of work is needed but it is a start. Contributions are appreciated. https://github.com/BytesRobotics/ros2_t260/tree/development

bobgates, 2020-07-27:
Also having the same problem. On Ubuntu 18.04, ROS2 foxy, librealsense built from branch refactor today (27/7/2020)

With a launch file edited from the `rs_multiple_devices.launch.py` down to:
```
def generate_launch_description():
    # config the serial number and base frame id of each camera
    camera2_base_frame_id = LaunchConfiguration('base_frame_id', default='camera2_link')
    camera2_serial_no = LaunchConfiguration('serial_no', default='905312110841')

    camera2_node = Node(
        package='realsense_node',
        executable='realsense_node',
        namespace="/camera2",
        output='screen',
        parameters=[{'serial_no':camera2_serial_no,
                    'base_frame_id': camera2_base_frame_id}]
def generate_launch_description():
    # config the serial number and base frame id of each camera
    camera2_base_frame_id = LaunchConfiguration('base_frame_id', default='camera2_link')
    camera2_serial_no = LaunchConfiguration('serial_no', default='905312110841')

    camera2_node = Node(
        package='realsense_node',
        executable='realsense_node',
        namespace="/camera2",
        output='screen',
        parameters=[{'serial_no':camera2_serial_no,
                    'base_frame_id': camera2_base_frame_id}]def generate_launch_description():
    # config the serial number and base frame id of each camera
    camera2_base_frame_id = LaunchConfiguration('base_frame_id', default='camera2_link')
    camera2_serial_no = LaunchConfiguration('serial_no', default='905312110841')

    camera2_node = Node(
        package='realsense_node',
        executable='realsense_node',
        namespace="/camera2",
        output='screen',
        parameters=[{'serial_no':camera2_serial_no,
                    'base_frame_id': camera2_base_frame_id}]
        )
return launch.LaunchDescription([camera2_node])
```

I see nearly the same errors as the OP:
```
[INFO] [launch]: All log files can be found below /home/xilinx/.ros/log/2020-07-27-15-21-29-362586-pynq-13752
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense_node-1]: process started with pid [13755]
[realsense_node-1] [INFO] [1595863292.851817726] [camera2.camera]: Device with serial number 905312110841 was found.
[realsense_node-1] [INFO] [1595863292.855532979] [camera2.camera]: Create a node for T265 Camera
[realsense_node-1] 2020-07-27 15:21:32,985 WARNING [default] Could not open device Unable to open device interface
[realsense_node-1] 2020-07-27 15:21:32,993 WARNING [default] Could not open device Unable to open device interface
[realsense_node-1] 2020-07-27 15:21:33,001 WARNING [default] Could not open device Unable to open device interface
[realsense_node-1] terminate called after throwing an instance of 'rs2::error'
[realsense_node-1]   what():  No device connected
[ERROR] [realsense_node-1]: process has died [pid 13755, exit code -6, cmd '/home/xilinx/ros2_ws/install/realsense_node/lib/realsense_node/realsense_node --ros-args --params-file /tmp/launch_params_o9adwc1p -r __ns:=/camera2'].
```

In my case, this is another attempt in a different way to get the T265 to work reliably. @Michael-Equi, I'll give your package a try!

robotsoft, 2020-07-30:
I fixed the problem according to https://github.com/akirayou/ros2_intel_realsense/commit/6d9c8097e213994d27f67fbff3cd7cc6d3a7dbfd.

sharronliu, 2020-07-30:
Same comment as https://github.com/intel/ros2_intel_realsense/issues/135

gftrobots, 2020-08-13:
Dear all, it seems that the issue has not been fixed yet and workarounds need to be applied.
We are facing the same issue when launching the T265 node.

Could you please be so kind and summarize the current status:

- which branch to apply
- which fixes / changes to manually apply 
- snipped of launch file

We are using:
- Ubuntu 18.04
- ROS2 Dashing
- librealsense 2.36