intel/ros2_intel_realsense #147: Realsense node crashes when subscribe to pointcloud

justinIRBT, 2020-08-03:
I am running the realsense node with a d435 on a Raspberry Pi 4 using the latest OS based on Debian Buster and ROS2 foxy.  I am able to view the depth and RGB images on rviz2, but when I subscribe to the pointcloud, the node crashes.  I get the following backtrace:

0xb5d5b850 in librealsense::align::create_aligned_profile (this=0xaeb25b64, original_profile=..., 
    to_profile=...) at /root/librealsense/src/proc/align.cpp:167
167	/root/librealsense/src/proc/align.cpp: No such file or directory.
(gdb) bt
#0  0xb5d5b850 in librealsense::align::create_aligned_profile (this=0xaeb25b64, 
    original_profile=..., to_profile=...) at /root/librealsense/src/proc/align.cpp:167
#1  0xb5d5bf84 in librealsense::align::allocate_aligned_frame (this=0xaeb25b64, source=..., 
    from=..., to=...) at /root/librealsense/src/proc/align.cpp:227
#2  0xb5d5c678 in librealsense::align::process_frame (this=0xaeb25b64, source=..., f=...)
    at /root/librealsense/src/proc/align.cpp:287
#3  0xb5d922f4 in librealsense::generic_processing_block::<lambda(rs2::frame, const rs2::frame_source&)>::operator()(rs2::frame, const rs2::frame_source &) const (__closure=0xaeb0bfbc, f=..., 
    source=...) at /root/librealsense/src/proc/synthetic-stream.cpp:75
#4  0xb5d9d728 in rs2::frame_processor_callback<librealsense::generic_processing_block::generic_processing_block(char const*)::<lambda(rs2::frame, const rs2::frame_source&)> >::on_frame(rs2_frame *, rs2_source *) (this=0xaeb0bfb8, f=0xa395e048, source=0xaeb09338)
    at /root/librealsense/build/../include/librealsense2/hpp/rs_processing.hpp:128
#5  0xb5d91edc in librealsense::processing_block::invoke (this=0xaeb25b64, f=...)
    at /root/librealsense/src/proc/synthetic-stream.cpp:43
#6  0xb5b0e8f8 in rs2_process_frame (block=0xaeb20858, frame=0xa395e048, error=0x9db224a8)
    at /root/librealsense/src/rs.cpp:2073
#7  0xb5ac552e in rs2::filter::process(rs2::frame) const ()
   from /home/pi/install/lib/librealsense_ros.so
#8  0xb5ae3f4c in realsense::RealSenseD435::publishTopicsCallback(rs2::frame const&) ()
   from /home/pi/install/lib/librealsense_ros.so

aligned_profile and aligned_profile->get() are valid pointers, but aligned_profile->get()->profile is a nullptr.

alsora, 2020-08-04:
I tried to fix the problem by changing the stream profiles as mentioned in [1], but without success
This is what I tried:

I modified the `STREAM_FORMAT` map at `realsense_ros/include/realsense/rs_constants.hpp`to use `RS2_FORMAT_RGBA8` for the color stream.
I modified the `setupStream()` function at `realsense_ros/src/rs_base.cpp` to default initialize streams using 
```
    cfg.enable_stream(RS2_STREAM_DEPTH);
    cfg.enable_stream(RS2_STREAM_COLOR, RS2_FORMAT_RGBA8);
```

[1] https://github.com/IntelRealSense/librealsense/issues/2567

doronhi, 2020-08-04:
Intel's plan is to release a new ROS2 wrapper, based on the ROS1 wrapper.
Currently the version supports Eloquent distribution and is still missing some features, such as support for parameters changing during runtime. I also didn't try it out on Raspberry Pi. 
It is, however, more up to date. It supports the latest librealsense2 and has more features included.
The release date is not set yet. It's about 2 month from now. 
Maybe you would like to try it out. If it didn't work, at least we'll know that something is wrong in what we conceive as our future version and put the effort to fix it before the first release.
Currently, the eloquent version resides here: https://github.com/IntelRealSense/realsense-ros/tree/eloquent

justinIRBT, 2020-08-04:
Thanks for the response, we were using the "refactor" branch, which seemed to be the most recently active.  We can try with the "eloquent" branch instead.

doronhi, 2020-08-05:
> Thanks for the response, we were using the "refactor" branch, which seemed to be the most recently active. We can try with the "eloquent" branch instead.

@justinIRBT ,I meant, branch eloquent in the ROS1 repository: https://github.com/IntelRealSense/realsense-ros/tree/eloquent
I was not aware of the "refacotr" branch here and will take a good look into it.

alsora, 2020-08-07:
@doronhi We succesfully built and run the eloquent branch on a Raspberry Pi 4 wiht ROS 2 Foxy.

In this branch the default is that the pointcloud message is not published. Do you have instructions for enabling it?

doronhi, 2020-08-08:
add the following parameter: `filters:=pointcloud`
I think it's in the README.md in the "development" branch.

justinIRBT, 2020-08-09:
I was able to run with:
`ros2 run realsense2_camera realsense2_camera --ros-args -p filters:=pointcloud`
I see it say that pointcloud is enable in the startup printouts (see console_output.txt)
I get a lot of error printouts after initialization along the lines of:
`[INFO] [1596930202.586897798] [RealSenseCameraNode]: RealSense Node Is Up!
 08/08 16:43:22,728 WARNING [2819592080] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 08/08 16:43:22,904 ERROR [2742640528] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
[WARN] [1596930204.059433324] [RealSenseCameraNode]: Hardware Notification:IR stream start failure,1.59693e+12,Error,Hardware Error
[WARN] [1596930205.080222710] [RealSenseCameraNode]: Hardware Notification:USB REC overflow,1.59693e+12,Error,Hardware Error
 08/08 16:43:25,150 ERROR [2707259280] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130`

I see the topic /depth/color/points, when I subscribe to it on rviz2, nothing shows up and I see increased console printouts along the lines of:
`08/08 16:45:00,113 WARNING [2707259280] (messenger-libusb.cpp:30) reset_endpoint returned error, index: 130, error: Connection timed out, number: 110
 08/08 16:45:05,153 WARNING [2742640528] (messenger-libusb.cpp:30) reset_endpoint returned error, index: 132, error: Connection timed out, number: 110
 08/08 16:45:05,153 WARNING [2819592080] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/08 16:45:05,153 WARNING [2819592080] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/08 16:45:05,213 WARNING [2819592080] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/08 16:45:05,214 WARNING [2819592080] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/08 16:45:05,274 WARNING [2819592080] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`

When I killed the camera node and restarted it, it couldn't find the realsense anymore (when I disconnected/reconnected the cable it was fine again).  I attached the console_output and dmesg.  Any theories about what might be going on / alternate configurations to try?  Thanks for all your help!

[console_output.txt](https://github.com/intel/ros2_intel_realsense/files/5046596/console_output.txt)
[dmesg.txt](https://github.com/intel/ros2_intel_realsense/files/5046597/dmesg.txt)

doronhi, 2020-08-09:
Most of the errors you receive comes from using the RSUSB backend. Where did you get your librealsense2 from? If it was built from source you could try change the CMake flag FORCE_RSUSB_BACKEND to false.
If you installed it from ros-<distro>-librealsense2 the try uninstalling it and install according to the librealsense2 guide: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md.

justinIRBT, 2020-08-09:
We build from source, I re-built with:
` cmake   -D CMAKE_INSTALL_PREFIX:PATH=/usr   -D CMAKE_BUILD_TYPE=“Release” -D BUILD_SHARED_LIBS=OFF   -D FORCE_RSUSB_BACKEND=OFF -D BUILD_PYTHON_BINDINGS=OFF -D BUILD_EXAMPLES=OFF -D BUILD_GRAPHICAL_EXAMPLES=OFF -D BUILD_GLSL_EXTENSIONS=OFF   ..`
Then I don't get usb messages when I run, but I am still not getting data in my pointcloud subscription when I run with filters:=pointcloud
I see the following on the command-line:
```
[ERROR] [1597001702.529708586] [RealSenseCameraNode]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
 09/08 12:35:03,582 WARNING [2760769424] (ds5-timestamp.cpp:73) UVC metadata payloads not available. Please refer to the installation chapter for details.
 09/08 12:35:03,600 ERROR [2760769424] (backend-v4l2.h:192) xioctl(VIDIOC_QBUF) guard failed for fd 22
 09/08 12:35:03,600 WARNING [2760769424] (backend-v4l2.h:195) xioctl(VIDIOC_QBUF) Re-enqueue succeeded for fd 22
 09/08 12:35:03,600 WARNING [2760769424] (backend-v4l2.h:199) xioctl(VIDIOC_QBUF) re-deque succeeded for fd 22
 09/08 12:35:03,600 ERROR [2760769424] (backend-v4l2.h:192) xioctl(VIDIOC_QBUF) guard failed for fd 22
 09/08 12:35:03,600 WARNING [2760769424] (backend-v4l2.h:195) xioctl(VIDIOC_QBUF) Re-enqueue succeeded for fd 22
 09/08 12:35:03,601 WARNING [2760769424] (backend-v4l2.h:199) xioctl(VIDIOC_QBUF) re-deque succeeded for fd 22
 09/08 12:35:03,601 ERROR [2760769424] (backend-v4l2.h:192) xioctl(VIDIOC_QBUF) guard failed for fd 22
 09/08 12:35:03,601 WARNING [2760769424] (backend-v4l2.h:202) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 22
 09/08 12:35:03,601 ERROR [2760769424] (backend-v4l2.h:192) xioctl(VIDIOC_QBUF) guard failed for fd 22
 09/08 12:35:03,601 WARNING [2760769424] (backend-v4l2.h:202) xioctl(VIDIOC_QBUF) Re-enqueue failed for fd 22
 09/08 12:35:03,629 ERROR [2760769424] (backend-v4l2.cpp:368) Non-sequential Video and Metadata v4l buffers
 09/08 12:35:03,629 WARNING [2760769424] (backend-v4l2.cpp:971) Video frame dropped, video and metadata buffers inconsistency
```
I am guessing the next step is to apply some of the patches for the uvcvideo module, as described here: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md.  We are running Raspbian Buster (kernel 5.4.51-v7l+), so I am assuming I will need to apply the patches by hand for the relevant files.
For arch, I see:
```
patch -p1 < ../../realsense-camera-formats.patch
patch -p1 < ../../realsense-metadata-bionic-5.patch
```

For ubuntu, I see:
```
patch -p1 < ../scripts/realsense-camera-formats-${ubuntu_codename}-${kernel_branch}.patch
patch -p1 < ../scripts/realsense-metadata-${ubuntu_codename}-${kernel_branch}.patch
patch -p1 < ../scripts/realsense-hid-${ubuntu_codename}-${kernel_branch}.patch
patch -p1 < ../scripts/realsense-powerlinefrequency-control-fix.patch
```

Which patches should I focus on to get the pointcloud publishing?

justinIRBT, 2020-08-10:
I manually patched the uvcvideo.ko module as described here:
https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense#Patching_uvcvideo
with the following patches:
https://github.com/IntelRealSense/librealsense/blob/master/scripts/realsense-camera-formats-bionic-5.patch
https://github.com/IntelRealSense/librealsense/blob/master/scripts/realsense-metadata-bionic-5.patch

I am still not getting a point cloud publishing.
When I plug the camera in, dmesg shows
```
[  +5.280228] usb 2-2: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[  +0.031096] usb 2-2: New USB device found, idVendor=8086, idProduct=0b07, bcdDevice=50.b1
[  +0.000018] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000016] usb 2-2: Product: Intel(R) RealSense(TM) Depth Camera 435 
[  +0.000016] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435 
[  +0.000014] usb 2-2: SerialNumber: 015223021316
[  +0.005336] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[  +0.005840] uvcvideo 2-2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[  +0.000022] uvcvideo 2-2:1.0: Entity type for entity Processing 2 was not initialized!
[  +0.000019] uvcvideo 2-2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[  +0.000018] uvcvideo 2-2:1.0: Entity type for entity Camera 1 was not initialized!
[  +0.000477] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.0/input/input1
[  +0.001556] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[  +0.004521] uvcvideo 2-2:1.3: Entity type for entity Processing 7 was not initialized!
[  +0.000021] uvcvideo 2-2:1.3: Entity type for entity Extension 8 was not initialized!
[  +0.000018] uvcvideo 2-2:1.3: Entity type for entity Camera 6 was not initialized!
[Aug10 15:52] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  +0.050402] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
```

The realsense driver start shows
```
$ ros2 run realsense2_camera realsense2_camera --ros-args -p filters:=pointcloud
[INFO] [1597100366.427047510] [RealSenseCameraNode]: RealSense ROS v2.2.14
[INFO] [1597100366.427595186] [RealSenseCameraNode]: Running with LibRealSense v2.36.0
[INFO] [1597100366.479407224] [RealSenseCameraNode]: Device with serial number 020522070648 was found.

[INFO] [1597100366.479750162] [RealSenseCameraNode]: Device with physical ID /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.0/video4linux/video0 was found.
[INFO] [1597100366.479897382] [RealSenseCameraNode]: Device with name Intel RealSense D435 was found.
[INFO] [1597100366.481239173] [RealSenseCameraNode]: Device with port number 2-2 was found.
print0
print1
print2
print3
[INFO] [1597100366.494534917] [RealSenseCameraNode]: getParameters...
[INFO] [1597100366.495372699] [RealSenseCameraNode]: setupDevice...
[INFO] [1597100366.495464067] [RealSenseCameraNode]: JSON file is not provided
[INFO] [1597100366.495690971] [RealSenseCameraNode]: Device Name: Intel RealSense D435
[INFO] [1597100366.496259553] [RealSenseCameraNode]: Device Serial No: 020522070648
[INFO] [1597100366.496398180] [RealSenseCameraNode]: Device physical port: /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.0/video4linux/video0
[INFO] [1597100366.496600066] [RealSenseCameraNode]: Device FW version: 05.11.01.100
[INFO] [1597100366.496755285] [RealSenseCameraNode]: Device Product ID: 0x0B07
[INFO] [1597100366.496881486] [RealSenseCameraNode]: Enable PointCloud: On
[INFO] [1597100366.497032761] [RealSenseCameraNode]: Align Depth: Off
[INFO] [1597100366.497179537] [RealSenseCameraNode]: Sync Mode: On
[INFO] [1597100366.497380644] [RealSenseCameraNode]: Device Sensors: 
[INFO] [1597100366.528799954] [RealSenseCameraNode]: Stereo Module was found.
[INFO] [1597100366.630044110] [RealSenseCameraNode]: RGB Camera was found.
[INFO] [1597100366.630207811] [RealSenseCameraNode]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[INFO] [1597100366.630308050] [RealSenseCameraNode]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[INFO] [1597100366.630391456] [RealSenseCameraNode]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[INFO] [1597100366.630474158] [RealSenseCameraNode]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[INFO] [1597100366.630575545] [RealSenseCameraNode]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[INFO] [1597100366.630688006] [RealSenseCameraNode]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[INFO] [1597100366.630795282] [RealSenseCameraNode]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[INFO] [1597100366.630929502] [RealSenseCameraNode]: Add Filter: pointcloud
[INFO] [1597100366.639715882] [RealSenseCameraNode]: num_filters: 1
[INFO] [1597100366.642440815] [RealSenseCameraNode]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[INFO] [1597100366.643595850] [RealSenseCameraNode]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[INFO] [1597100366.644670738] [RealSenseCameraNode]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[... cut ...]
```[WARN] [1597100977.984372341] [RealSenseCameraNode]: No stream match for pointcloud chosen texture Process - Color
 10/08 16:09:39,070 WARNING [2760888208] (ds5-timestamp.cpp:73) UVC metadata payloads not available. Please refer to the installation chapter for details.
 10/08 16:09:39,081 ERROR [2760888208] (backend-v4l2.h:192) xioctl(VIDIOC_QBUF) guard failed for fd 22
 10/08 16:09:39,081 WARNING [2760888208] (backend-v4l2.h:195) xioctl(VIDIOC_QBUF) Re-enqueue succeeded for fd 22
 10/08 16:09:39,081 WARNING [2760888208] (backend-v4l2.h:199) xioctl(VIDIOC_QBUF) re-deque succeeded for fd 22
```

justinIRBT, 2020-08-11:
I ran some more experiments on a raspberry pi 4 with the patched uvcvideo module powered from the raspberry pi power adapter to eliminate power issues as an error source.  I saw in another thread (https://github.com/IntelRealSense/realsense-ros/issues/715) where someone was getting
```
[ERROR] [1597187329.327184098] [RealSenseCameraNode]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
```
you mentioned `enable_infra2:=false`

I tried running with
```
$ ros2 run realsense2_camera realsense2_camera --ros-args -p filters:=pointcloud -p enable_infra2:=false -p enable_infra1:=false -p enable_depth:=false
```

I no longer see the "error has occurred during frame callback", but I still don't get any publishing on the /depth/color/points topic.

I also tried with depth enabled, but the two infra off.

Some things I see in dmesg
```
[  +0.001275] uvcvideo: Non-zero status (-71) in video completion handler.
[  +0.196610] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
```

Any suggestions of more things to try to get the point cloud publication?

doronhi, 2020-08-12:
For pointcloud with color texture you need both color and depth. You don't need infra1 and infra2.

> I see the "No stream match for pointcloud chosen texture" message when I subscribe to the pointcloud from rviz, but I saw on another forum that the error could be ignored.
Could you refer me to the link please? I would like to see the overall connection.

Assuming you can see in RViz (or other node) the video from both depth and color coming ok, the next think that comes to mind is that maybe it's a synchronization issue. Could you check the time tags? Run `ros2 topic echo /depth/image_rect_raw --no-arr` and `ros2 topic echo /color/image_raw --no-arr` (together, in different terminals). Do the time stamps seems similar? Could you post a sample of each?

justinIRBT, 2020-08-12:
For the "No stream match for pointcloud chosen texture Process - Color", I was referring to this thread (https://github.com/IntelRealSense/realsense-ros/issues/588)

I ran:
```
ros2 run realsense2_camera realsense2_camera --ros-args -p filters:=pointcloud -p enable_infra2:=false -p enable_infra1:=false
```

I was able to view color and depth stream in rviz.  I attached a side by side of the 2 topics echos you requested, the timestamps are not an exact match.
[image]

doronhi, 2020-08-13:
The timestamps show that the images are published every 110 millisecond (around 9Hz), instead of the expected rate of 30Hz (based on the log you published previously).
Could your machine be overloaded? Could you try reducing the frame rate to 6Hz? (`color_fps:=6 depth_fps:=6 enable_infra1:=false enable_infra2:=false`)

justinIRBT, 2020-08-13:
I attached an image showing top with the 6 FPS config, the system is pretty lightly loaded, but I am not seeing the pointcloud publishing.  When I don't set the FPS, I see the system at 66% idle on top.
[image]

justinIRBT, 2020-08-14:
Interestingly, I was randomly testing something else and a single pointcloud publication made it through and displayed on rviz2.  I was running without any of the throttling parameters, just filters:=pointcloud.  I was also subscribed to the depth and color images.  I waved my hand in front of it after the point cloud showed up, but it never updated.  The color and depth images were showing my hand moving.  I tried topic hz after I saw it and it never published anything.  I wasn't able to reproduce after a couple of restarts.

justinIRBT, 2020-08-18:
I made an executable based off of rs-pointcloud just using librealsense2 (I am running headless, so I can't use the example off the shelf.)

My code is:
```
int main(int argc, char * argv[]) try
{

    // Declare pointcloud object, for calculating pointclouds and texture mappings
    rs2::pointcloud pc;
    // We want the points object to be persistent so we can display the last cloud when a frame drops
    rs2::points points;

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    // Start streaming with default recommended configuration
    pipe.start();

    while (true) // Application still alive?
    {
        // Wait for the next set of frames from the camera
        auto frames = pipe.wait_for_frames();

        auto color = frames.get_color_frame();

        // For cameras that don't have RGB sensor, we'll map the pointcloud to infrared instead of color
        if (!color)
            color = frames.get_infrared_frame();

        // Tell pointcloud object to map to this color frame
        pc.map_to(color);

        auto depth = frames.get_depth_frame();

        // Generate the pointcloud and texture mappings
        points = pc.calculate(depth);
        auto t1 = std::chrono::high_resolution_clock::now();
        auto duration = t1.time_since_epoch();
        auto millis = std::chrono::duration_cast<std::chrono::milliseconds>(duration).count();
        std::cerr << "Got points "<<points.size()<<" at time "<<millis<<std::endl;;
    }

    return EXIT_SUCCESS;
}
```

It prints:
```
Got points 407040 at time 1597778763248
Got points 407040 at time 1597778763356
Got points 407040 at time 1597778763463
Got points 407040 at time 1597778763571
Got points 407040 at time 1597778763678
Got points 407040 at time 1597778763785
Got points 407040 at time 1597778763892
Got points 407040 at time 1597778764002
Got points 407040 at time 1597778764109
Got points 407040 at time 1597778764217
Got points 407040 at time 1597778764324
Got points 407040 at time 1597778764431
```

Which is about the 9+ Hz rate we were seeing the image frames come in at when we didn't throttle the FPS.  I am assuming this implies the device is working properly on my Raspberry Pi setup and the issue is somewhere in the ROS driver implementation?