intel/ros2_intel_realsense #151: D435i imu data not published

rjshim, 2020-08-18:
Hi, 

I tested below with a realsense D435i model and it doesn't show any results.
$ ros2 launch realsense_ros2_camera ros2_intel_realsense.launch.py
$ ros2 topic echo /camera/accel/sample
$ ros2 topic echo /camera/gyro/sample
I assume the realsense_ros2_camera node is not publishing any messages.

Thank you in advance,
Ryan

rafaelRinconFo, 2020-09-25:
hI!! I still have this issue, did you find any solution?

JaehyunShim, 2020-10-16:
@rafaelRinconFo No, unfortunately.

JaehyunShim, 2020-10-16:
ping @ahuizxc @sharronliu @yechun1

rafaelRinconFo, 2020-10-16:
> @rafaelRinconFo No, unfortunately. 

Did you used another Ros2 distribution? Did you just simply let the IMU data without being used ?

JaehyunShim, 2020-10-16:
@rafaelRinconFo 

For now, my D435i is useless. I am just shocked by their customer service for a 200$ device 

https://store.intelrealsense.com/?_ga=2.66320430.1073046613.1602814530-290743533.1597819980

rafaelRinconFo, 2020-10-26:
@JaehyunShim  It seems to be that all the Intel Support for ROS2 was moved from Dashing to Eloquent and Foxy in this repo 
[https://github.com/IntelRealSense/realsense-ros](url)

JaehyunShim, 2020-10-27:
@rafaelRinconFo 

They never worked on Dashing, but they just started with Eloquent directly. They do not have enough people to work on ROS packages, so you might need to wait another half a year or something to see properly working ROS2 support for Intel Realsense. I recently sent a PR to help a bit (https://github.com/IntelRealSense/realsense-ros/commits/eloquent) 

To be honest, I do not even understand why the main developer guy on the list (it seems he is the only one working on the repo you linked) doesn't work on this repo, but created his own repo and is trying to do everything from scratch on his own. If you want to see the progress of ROS support for Realsense, keep tracking [this](https://github.com/IntelRealSense/librealsense/issues/5825).

closed 2020-10-28 by JaehyunShim

ahuizxc, 2020-10-28:
> @rafaelRinconFo
>
> They never worked on Dashing, but they just started with Eloquent directly. They do not have enough people to work on ROS packages, so you might need to wait another half a year or [...]

sorry man, I am pretty sure the last test before i am leaving this team was passed :(, maybe you can try the older version?

ryanmaxwell96, 2022-12-08:
I cannot get imu data to publish either, even though it clearly works when looking from realsense-viewer: Jetson AGX Xavier, Ubuntu 20.04, Jetpack 5.0.2, Foxy