intel/ros2_intel_realsense #155: D455 Camera - No RealSense devices were found

sangohku, 2020-09-04:
I am having an issue getting the ROS2 realsense wrapper working for the D455 where it can't find the device. 

Running 
`. /opt/ros/dashing/setup.bash
ros2 run realsense_ros2_camera realsense_ros2_camera
`
Gives the following result:
[INFO] [RealSenseCameraNode]: RealSense ROS v2. 0.1
[INFO] [RealSenseCameraNode]: Running with LibRealSense v2.16.5
[INFO] [RealSenseCameraNode]: getParameters...
[INFO] [RealSenseCameraNode]: setupDevice...
[ERROR] [RealSenseCameraNode]: No RealSense devices were found! Terminate RealSense Node...

My system:
Jetson Nano
Ubuntu 18.04.5 LTS
ROS Dashing

`realsense-viewer` is working fine
`rs-enumerate-devices`  gives the following

Device info: 
    Name                          : 	Intel RealSense D455
    Serial Number                 : 	030422250601
    Firmware Version              : 	05.12.07.100
    Recommended Firmware Version  : 	05.12.06.00
    Physical Port                 : 	2-1.1-4
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B5C
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	025523050418
    Firmware Update Id            : 	025523050418

Can anyone help with getting troubleshooting this issue?

TSC21, 2020-09-10:
Same problem here.

JohnBetaCode, 2020-10-14:
the same problem here on master branch, using dashing!

rafaelRinconFo, 2020-10-14:
I'm experiencing the same Issue with dashing.

ipirva, 2021-03-26:
I am experiencing the same error with Intel RealSense D455
`Device info:
    Name                          : 	Intel RealSense D455
    Serial Number                 : 	xxx
    Firmware Version              : 	05.12.10.00
    Recommended Firmware Version  : 	05.12.10.00
    Physical Port                 : 	/sys/devices/3610000.xhci/usb2/2-3/2-3.1/2-3.1:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B5C
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	049423060026
    Firmware Update Id            : 	049423060026`

`ros2 run realsense_ros2_camera realsense_ros2_camera
[INFO] [RealSenseCameraNode]: RealSense ROS v2. 0.1
[INFO] [RealSenseCameraNode]: Running with LibRealSense v2.16.5
[INFO] [RealSenseCameraNode]: getParameters...
[INFO] [RealSenseCameraNode]: setupDevice...
[ERROR] [RealSenseCameraNode]: No RealSense devices were found! Terminate RealSense Node...`

uoyssim, 2021-05-11:
I'm experiencing the same issue.

Ubuntu 18.04.5 LTS
ROS2 Dashing

I tested *realsense-viewer*

It works fine until I run
```
. /opt/ros/dashing/setup.bash
```
After I run setup.bash, It can't find realsense devices.

**lsusb -v**
```
Bus 002 Device 004: ID 8086:0b5c Intel Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x8086 Intel Corp.
  idProduct          0x0b5c 
  bcdDevice           50.cb
  iManufacturer           1 Intel(R) RealSense(TM) Depth Camera 455 
  iProduct                2 Intel(R) RealSense(TM) Depth Camera 455 
  iSerial                 3 028223051268
  bNumConfigurations      1
...
```