intel/ros2_intel_realsense #162: colcon build --symlink-install with error

Schloern93, 2020-09-22:
Hello,
I'm trying to get the new intel Real Sense Tracking Camera T265 running. I have tried to build the provided package according to the manual. 
I get the following error message in the terminal. It is attached to the message. Ask for help.
I used Ubuntu 18.04 and installed ros-dashing-desktop.

Thanks and best regards 
Florian

[Error Msg.txt](https://github.com/intel/ros2_intel_realsense/files/5263084/Error.Msg.txt)

shunta3296, 2020-10-30:
Hello,
I have the exact same problem.
The errors are the same and the environment is the same.
Have you found a solution to this problem?

Schloern93, 2020-10-30:
Hey,

Yeah i found a way which works. Just Remind me on Soneday. Im not at home for a few days

Von: shunta3296
Gesendet: Freitag, 30. Oktober 2020 10:53
An: intel/ros2_intel_realsense
Cc: Schloern93; Author
Betreff: Re: [intel/ros2_intel_realsense] colcon build --symlink-install with error (#162)

Hello,
I have the exact same problem.
The errors are the same and the environment is the same.
Have you found a solution to this problem?

ShuntaIto-keio, 2020-11-01:
Hi,

I'm really glad to get your reply.
Japan is now on Sunday. Is it Sunday in your country already?
Please let me know how you solved that problem.

AndreaEirale, 2021-01-05:
Hi everybody! 
I am experiencing the exact same error! Can you please let me know how you solved the problems?
Thank you very much!

Schloern93, 2021-01-06:
Hallo,
I have updated my system to ros eloquent. And switched to the current project from intel: 
https://github.com/IntelRealSense/realsense-ros

Schloern93, 2021-01-08:
I close this issue now.

closed 2021-01-08 by Schloern93