intel/ros2_intel_realsense #171: ros foxy installation

dbgarasiya, 2020-11-26:
Hello,

I want to install foxy(ROS) on Ubuntu 20.04.1 LTS
I was unable to install and got error while :  colcon build --symlink-install

CMake Error at CMakeLists.txt:19 (find_package):

Failed   <<< realsense_msgs [1.73s, exited with code 1]

Summary: 0 packages finished [2.49s]
  1 package failed: realsense_msgs
  1 package had stderr output: realsense_msgs
  3 packages not processed

colcon build failed:realsense_msgs

zmk5, 2021-03-12:
Check the `CMakeLists.txt` file for `realsense_msgs` and see if you have all the dependencies required by it installed. It seems like you are likely missing one of them.

cardboardcode, 2021-04-05:
Hi @dbgarasiya, 

If you still encountering this issue, you can follow the installation instructions below:
```bash 
# Download Realsense SDK 2.0
sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo focal main" -u
sudo apt-get install librealsense2-dkms librealsense2-utils librealsense2-dev librealsense2-dbg -y
# Create a ROS2 workspace
mkdir -p ~/camera_ws/src && cd ~/camera_ws/src
# Download and Set-up ros2_intel_realsense package.
git clone https://github.com/intel/ros2_intel_realsense --single-branch --branch refactor --depth 1
cd ~/camera_ws
colcon build
source install/setup.bash
# Launch It
ros2 run realsense_node realsense_node
```

Hope this helps whoever is still encountering this issue.

ThotAlion, 2021-04-30:
Hello, thank you for the installation process. I did try this process but without success. Here below are the details : 
- Ubuntu 20.04
- ROS2 foxy base
- Hardware is an Upboard
- T265 connected on USB2.1

The realsense viewer works.

the rs-enumerate-devices gives : 
```
Device info:
    Name                          :     Intel RealSense T265
    Serial Number                 :     943XXXXXXX78
    Firmware Version              :     0.2.0.951
    Physical Port                 :     1-7.1-12
    Product Id                    :     0B37
    Usb Type Descriptor           :     2.1
    Product Line                  :     T200

Stream Profiles supported by Tracking Module
 Supported modes:
    stream       resolution      fps       format
    Fisheye 1     848x800       @ 30Hz     Y8
    Fisheye 2     848x800       @ 30Hz     Y8
    Gyro         N/A            @ 200Hz    MOTION_XYZ32F
    Accel        N/A            @ 62Hz     MOTION_XYZ32F
    Pose         N/A            @ 200Hz    6DOF
```
But after setting the right serial number in t265.yaml, I have the following error : 

```
upboard:~/camera_ws$ ros2 run realsense_node realsense_node
[INFO] [1619782958.462440634] [camera]: Device's serial number is not set, enabling the default device!
[INFO] [1619782958.533307579] [camera]: Device with serial number 943222111378 was found.
[INFO] [1619782958.533632005] [camera]: Create a node for T265 Camera
terminate called after throwing an instance of 'rs2::error'
  what():  No device connected

```

I did the same test on a standard desktop with the same configuration, and we have same error.

Have you heard about this kind of issue (realsense camera on ros2 foxy) ?

Thanks,

cardboardcode, 2021-05-01:
@ThotAlion For my case, my **Hardware** is a `Intel Realsense D415`. 

This issue may be specific to this model series of `T265`. Unfortunately, I have not tried this model yet. 

> Have you heard about this kind of issue (realsense camera on ros2 foxy) ?

I have encountered it myself when attempting to use `Intel Realsense D415` on **ROS2 Foxy**, using the previous source builds. However, I don't encounter this issue any longer by doing the instructions I recommended. 😅

cardboardcode, 2021-05-25:
@ThotAlion 
This project has been deprecated so have you tried the **realsense_ros**'s ros2 branch? Maybe they had added more recent updates for higher compatibility with the `T265` models. 

I have updated my previous instructions to use the code under **realsense_ros** instead. See below:
```bash
# Download Realsense SDK 2.0
sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo focal main" -u
sudo apt-get install librealsense2-dkms librealsense2-utils librealsense2-dev librealsense2-dbg -y
# Create a ROS2 workspace
mkdir -p ~/camera_ws/src && cd ~/camera_ws/src
# Download and Set-up ros2_intel_realsense package.
git clone --depth 1 --branch `git ls-remote --tags https://github.com/IntelRealSense/realsense-ros.git | grep -Po "(?<=tags/)3.\d+\.\d+" | sort -V | tail -1` https://github.com/IntelRealSense/realsense-ros.git
cd ~/camera_ws
colcon build
source install/setup.bash
# Launch It
ros2 launch realsense2_camera rs_launch.py enable_pointcloud:=true
```

zmk5, 2021-05-30:
Yeah this project is pretty much deprecated. I just noticed the `ros2` branch on `realsense_ros` which has been great so far.

ThotAlion, 2021-05-30:
@cardboardcode 
Yes, I saw the last release of the [realsense-ros](https://github.com/IntelRealSense/realsense-ros) on the 5th May 2021. I tested, and it works very well.
BTW, thank you for the new install process :-)