intel/ros2_intel_realsense #175: T265 on Jetson AGX Xavier with ROS2 Dashing

AndreaEirale, 2021-01-26:
I can't make T265 work on Jetson AGX Xavier with Ubuntu 18.04 and ROS2 Dashing, installing the Dashing branch from this repository.

The error I encounter is:
```
[INFO] [launch]: All log files can be found below /home/me/.ros/log/2021-01-26-16-54-34-349447-me-desktop-17650
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense_node-1]: process started with pid [17666]
[realsense_node-1] terminate called after throwing an instance of 'rs2::error'
[realsense_node-1]   what():  No device connected
[ERROR] [realsense_node-1]: process has died [pid 17666, exit code -6, cmd '/home/me/ros2_ws/install/realsense_node/lib/realsense_node/realsense_node __ns:=/'].
```
On my pc, running Ubuntu 18.04, I solved this problem by adding: `dev_.hardware_reset();` in function RealSenseBase::startPipeline() of `rs_base.cpp`, but the same workaround did not work on Jetson, probably for the different Kernel supported for Jetson which require a different RealSense SDK.

Can you help me with this problem?

SuperShinyEyes, 2021-03-30:
Hi @AndreaEirale. Have you fixed the issue?

AndreaEirale, 2021-04-07:
Hi @SuperShinyEyes 

I managed to make the camera work by reinstalling all RealSense libraries and making the changes written in my previous message.
However, now I have an even bigger problem. When I launch the rs_demo_t265.launch.py file rviz2 is automatically opened and I can see all TF correctly, including camera_pose_optical_frame and odom_frame. My problem is, if I close that rviz, all TF are not published anymore. So, if I open rviz again, I can't see the camera's TF. 
How is this possible? I have really never seen such a behavior in ROS2 in my whole life!
Can you please explain me what is happening?
Thank you very much!

AndreaEirale, 2021-04-08:
Ok, I managed to find the problem here. You have necessarily to add the voice "Odometry" in Rviz, subscribing to the topic "/camera/odom/sample". If this subscription is not done, the camera odometry is not published, and as a consequence also TFs are not published.

There are no other problems, I close the discussion

closed 2021-04-08 by AndreaEirale

SuperShinyEyes, 2021-04-13:
@AndreaEirale Thank you for the update. Glad you got it working :)